AlexandreRouma/SDRPlusPlus #260: [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig

JosephTheTank13, 2021-08-15:
I installed Pop OS on an old Dell workstation computer with old integrated graphics. The OS works ok but I haven't been able to update it yet since the System76 PPA servers have been offline for the last day. When I try to launch SDR++, it gives this error:
`[error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig`. I've seen that a graphics driver update could resolve this issue, and it may work once the server is back online. But I've installed it on two different computers from the same era with different gpus and they both threw the error. Any ideas on a solution? The error does not show up on my modern PC running the same OS.

waterwin, 2021-08-15:
Your graphics card might be too old to run OpenGL above version 2.1

JosephTheTank13, 2021-08-15:
Oh dear.. That's probably what it is. That sucks.

AlexandreRouma, 2021-08-15:
gonna work on supporting older GLs

JosephTheTank13, 2021-08-15:
Alright! Well I will check in occasionally and try new versions to see if they work. Thanks btw for this awesome program, It's really really good.

waterwin, 2021-08-15:
Sometimes there is only one way and that is forward.
Just found out that the Windows laptop also only has OpenGL 2.1, Intel driver already the latest.

JosephTheTank13, 2021-08-15:
What does that mean for Linux? I think it works on windows even on the old machines at this time

waterwin, 2021-08-16:
Define "old" ;-) and "it" and "that"

As far as I understand it: specific hardware needs a driver for the OS you want to use.
My laptops have Intel HD Graphics. Intel decided to not make newer Linux drivers for this hardware. Since many years ago.
Also specific software tricks in newer graphics software might need newer or faster hardware or hardware with a different architecture.
OpenGL in its latest versions uses all the new tricks developed lately. Not all tricks used by software developers integrating OpenGL will work with older versions of OpenGL and mutatis mutandis not with older drivers or older hardware.

Example: if your software or driver would only work with a dedicated floating point co-processor (FPU) it would not work with older cpu's not having that kind of extra.

AlexandreRouma, 2021-08-16:
Gonna need full startup logs on a non working system.
Trying to add a work around for this.

AlexandreRouma, 2021-08-16:
yeah, just run it in a terminal (adding the -s parameter if you're on windows) and send the full logs until the crash.
I wanna know exactly at which part of loading OpenGL is unhappy.

JosephTheTank13, 2021-08-16:
Terminal output: `$ sdrpp
[2021-08-16 17:32:32.112] [info] SDR++ v1.0.3
[2021-08-16 17:32:32.112] [info] Loading config
[2021-08-16 17:32:32.157] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
`

I know that's not super helpful but that's what it puts out.

AlexandreRouma, 2021-08-16:
it's enough for me to know where it happens.
Seems to be where I thought it was: when asking glfw for a specific opengl version.
I'll see what I can do about it.

JosephTheTank13, 2021-08-16:
Thank you so much for being responsive and looking into it!

Lesprilibre, 2021-08-17:
> gonna work on supporting older GLs

Thanks a lot !

AlexandreRouma, 2021-08-21:
I've (hopefully) added support for OpenGL 2.1 as well as OpenGL ES, could you try the latest commit and see if it runs properly?
(If you wait a few minutes from now, the automated build will be done so you can try that if you prefer)

waterwin, 2021-08-21:
Can you please advise from where to download latest automated build? I only find the releases which is same as before. Thank you.

JosephTheTank13, 2021-08-21:
I compiled from the source and this is what I get

`sdrpp`
`[2021-08-21 17:07:59.989] [info] SDR++ v1.0.4` 
`[2021-08-21 17:08:00.023] [info] Loading config`
`[2021-08-21 17:08:03.843] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig`
`[2021-08-21 17:08:03.916] [info] OpenGL 3.0 was not supported`
`[2021-08-21 17:08:04.038] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig`
`[2021-08-21 17:08:04.087] [info] OpenGL 3.1 ES was not supported`
`[2021-08-21 17:08:04.087] [error] Glfw Error 65540: Invalid OpenGL ES version 2.1`
`[2021-08-21 17:08:04.087] [info] OpenGL 2.1 was not supported`
`sdrpp: /build/glfw3-KLyziG/glfw3-3.3.2/src/window.c:1040: glfwSetWindowMaximizeCallback: Assertion window != NULL failed.`
`Aborted (core dumped)`

AlexandreRouma, 2021-08-21:
> I compiled from the source and this is what I get

My bad, looks like I forgot a line. I pushed a fix just now, try the latest commit

> Can you please advise from where to download latest automated build? I only find the releases which is same as before. Thank you.

Each commit is built and available through [the "Actions" tab](https://github.com/AlexandreRouma/SDRPlusPlus/actions)
At the time of writing this, the latest commit still has an orange sign next to it meaning it's compiling.
When it becomes a green checkmark, click on it and scroll down to the "Artifacts" section. You can then click on the file you want to download it

JosephTheTank13, 2021-08-21:
sdrpp
[2021-08-21 18:03:28.272] [info] SDR++ v1.0.4
[2021-08-21 18:03:28.273] [info] Loading config
[2021-08-21 18:03:32.636] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2021-08-21 18:03:32.736] [info] OpenGL 3.0 was not supported
[2021-08-21 18:03:32.782] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2021-08-21 18:03:32.793] [info] OpenGL 3.1 ES was not supported
[2021-08-21 18:03:32.839] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2021-08-21 18:03:32.840] [info] OpenGL 2.1 was not supported
sdrpp: /build/glfw3-KLyziG/glfw3-3.3.2/src/window.c:1040: glfwSetWindowMaximizeCallback: Assertion `window != NULL' failed.
Aborted (core dumped)

same error as before

AlexandreRouma, 2021-08-21:
This means your computer just doesn't support OpenGL, I'm afraid SDR++ won't work well for you in that case.

You *can* try software rendering with Mesa3D but expect very poor performance.

JosephTheTank13, 2021-08-21:
I ran  glxinfo | grep opengl but it didn't print anything and just returned the line. Was it supposed to say something as the output?

AlexandreRouma, 2021-08-21:
then you really don't have any OpenGL support at the moment.

Are you sure your graphics drivers are properly installed?

Could you give me info on your hardware

JosephTheTank13, 2021-08-21:
https://linux-hardware.org/?probe=4fa1d0db49

There's the hardware probe. Maybe it is just too old. :/ bummer

AlexandreRouma, 2021-08-21:
looks like it only supports opengl 1.4
Can't support that sorry, looks like it's too old.
Adding basically any GPU made in the last 10 years would fix this issue

JosephTheTank13, 2021-08-21:
Alright. Didn’t realize it was so old!

waterwin, 2021-08-22:
And works with my HP EliteBook 2540p Intel HD Graphics supporting OpenGL 2.1
Well done Sir and thank you! Only ran a short FM test with Pluto+.
Starting sdrpp -s first checks for OpenGL 3.0 then OpenGL 3.1 ES and then finds OpenGL 2.1

With cmake-gui I had to switch off airspyhf and airspyhf_source modules. A first for me to get so deep into the build system.