AlexandreRouma/SDRPlusPlus #261: Module instances are not deinitialized/stopped on exit

nanoant, 2021-08-16:
It looks that `sdrpp_main` is only calling `end` on all modules after it exits main runloop, but existing module instances are not de-initialized/released. As a side-effect e.g. `RTLSDRSourceModule` is not closing active device with `rtlsdr_close`, which in turn causes that ADCs and demodulators remain powered and the dongle is staying hot draining power even no app is using it.

I think this may be also an issue with `librtlsdr` itself because it can very likely happen that process just crashes or gets killed and we get the same problem with the dongle draining power for nothing. So I also plan to submit an issue there.

Anyways, I simply tried adding some new helper methods and calling
~~~
    // Shut down all instances and modules
    core::moduleManager.doDeleteAllInstances();
    core::moduleManager.doEndAllModules();
~~~
But then I get a crash here:
~~~
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff71b2333a libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff71bdfe60 libsystem_pthread.dylib`pthread_kill + 430
    frame #2: 0x00007fff71aaa808 libsystem_c.dylib`abort + 120
    frame #3: 0x00007fff6ed09458 libc++abi.dylib`abort_message + 231
    frame #4: 0x00007fff6ecfa7e9 libc++abi.dylib`demangling_terminate_handler() + 48
    frame #5: 0x00007fff6ed08887 libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #6: 0x00007fff6ed08838 libc++abi.dylib`std::terminate() + 56
    frame #7: 0x00007fff6ece55cd libc++.1.dylib`std::__1::thread::~thread() + 17
    frame #8: 0x00000001002865c4 libsdrpp_core.dylib`dsp::generic_block<dsp::FrequencyXlator<dsp::complex_t> >::~generic_block() + 180
    frame #9: 0x0000000100282bd6 libsdrpp_core.dylib`SignalPath::removeVFO(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) + 198
    frame #10: 0x000000010029022b libsdrpp_core.dylib`VFOManager::VFO::~VFO() + 379
    frame #11: 0x0000000100290a1d libsdrpp_core.dylib`VFOManager::deleteVFO(VFOManager::VFO*) + 413
    frame #12: 0x000000011308e9e6 radio.dylib`RadioModule::~RadioModule() + 166
    frame #13: 0x0000000113074f53 radio.dylib`_DELETE_INSTANCE_ + 19
    frame #14: 0x000000010027d8ca libsdrpp_core.dylib`ModuleManager::deleteInstance(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) + 234
    frame #15: 0x000000010027e61f libsdrpp_core.dylib`ModuleManager::doDeleteAllInstances() + 479
    frame #16: 0x00000001001615a3 libsdrpp_core.dylib`sdrpp_main(int, char**) + 26275
    frame #17: 0x00007fff719dbcc9 libdyld.dylib`start + 1
~~~

But if I just hardcode de-initialization of `RTLSDRSourceModule` then `rtlsdr_close` is called on exit and the dongle cools down after exit.

AlexandreRouma, 2021-08-16:
Not calling the instance destructor on close is on purpose to avoid taking a super long time to close.
Most SDR drivers handle the process exiting correctly, guess I'll have to check with librtlsdr

AlexandreRouma, 2021-08-16:
By the way, the crash you're experiencing is an unrelated bug when VFOs get deleted that somehow only happens to very few people and that I've never been able to locate. You should be able to get the same crash by deleting the radio module in the config manager.

nanoant, 2021-08-17:
> Not calling the instance destructor on close is on purpose to avoid taking a super long time to close.

Yeah I know that this is even recommended approach on UNIX systems as OS can cleanup stuff after the process (e.g. allocated memory) much faster than if the process would deallocate / deinitialize itself.

> Most SDR drivers handle the process exiting correctly, guess I'll have to check with librtlsdr

I think the problem here is that librtlsdr isn't really talking with the device via the driver, but rather it talks directly via libusb, so there's nothing (i.e. driver) in the OS to handle the processes' exit. Ideally we would want to issue USB reset command automatically when process ends, but not sure if this is something libusb or OS USB stack can trigger.

> By the way, the crash you're experiencing is an unrelated bug when VFOs get deleted that somehow only happens to very few people and that I've never been able to locate. You should be able to get the same crash by deleting the radio module in the module manager.

I'll to delete radio module there in module manager, and see if I get same crash.