AlexandreRouma/SDRPlusPlus #263: Compiling on Ubuntu 18.04 Bionic

craigerl, 2021-08-17:
In addition to following the Readme, I'm cmake'ing with these options trying to get this to compile on Ubuntu 18.04

cmake  -DCMAKE_CXX_FLAGS="-lstdc++fs" -D OPT_OVERRIDE_STD_FILESYSTEM=ON ..
make

Which ultimately results in,

```
In file included from /home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/dsp/resampling.h:6:0,
                 from /home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/dsp/vfo.h:4,
                 from /home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/signal_path/vfo_manager.h:2,
                 from /home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/gui/main_window.h:6,
                 from /home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/core.cpp:7:
/home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/dsp/math.h: In member function ‘int dsp::Add<T>::run()’:
/home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/dsp/math.h:67:17: error: there are no arguments to ‘volk_32fc_x2_add_32fc’ that depend on a template parameter, so a declaration of ‘volk_32fc_x2_add_32fc’ must be available [-fpermissive]
                 volk_32fc_x2_add_32fc((lv_32fc_t*)out.writeBuf, (lv_32fc_t*)_a->readBuf, (lv_32fc_t*)_b->readBuf, a_count);
                 ^~~~~~~~~~~~~~~~~~~~~
/home/craiger/ham/sdrplusplus/SDRPlusPlus/core/src/dsp/math.h:67:17: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
```

craigerl, 2021-08-17:
Additionally add this to the top of  core/CMakeLists.txt :

```
set(CMAKE_CXX_FLAGS "-fpermissive")
add_compile_options(-fpermissive)
```

cmake ignores the first line, but I put both.

craigerl, 2021-08-17:
Compiles on ubuntu 18.04 with above tweaks

closed 2021-08-17 by craigerl

AlexandreRouma, 2021-08-17:
Hello,
Good to know the OPT_OVERRIDE_STD_FILESYSTREM thing I added works, hadn't actually had a chance to try it out yet!

However, the issue with libvolk should be fixed differently. Ubuntu 18.04 is missing libvolk 1.4 which is the minimum requirement. 1.3 is sadly missing some critical functions that help with performance. (you'll notice stereo FM won't work)
I'll have to check if there are any other options to get it to build correction then -fpermissive

N0rbert, 2021-08-20:
Official deb-package for Ubuntu 18.04 LTS would be very useful. Or please provide [AppImage](https://github.com/AlexandreRouma/SDRPlusPlus/issues/286) for such systems.

N0rbert, 2021-08-20:
I have successfully compiled and ran SDR++ with RTL-SDR on Ubuntu 18.04 LTS, documented the [method on AskUbuntu](https://askubuntu.com/a/1359194/66509).

AlexandreRouma, 2021-08-20:
Without volk 1.4 FM stereo will not work

craigerl, 2021-08-22:
I've tested narrow and wide FM.  Audio becomes suddenly distorted now and then, but changing the audio bitrate (say from 44K to 22K) fixes it at least for a while.  The distortion sounds like low cpu or buffer drops.  You can make out the audio put it pulses at 10Hz or so.

I'll look into upgrading volk if it's not a cascading dependency nightmare on Ubuntu 18.04.

I'm not blaming sdrplusplus here, I'm surprised it works at all on this old distro.

craigerl, 2021-08-22:
I see the audio stuttering is solved in #112 , but after applying that fix, sdrpp doesn't start at all.

```
[2021-08-22 10:15:40.494] [info] Initializing Radio (radio)
[2021-08-22 10:15:40.556] [error] Could not open audio device
[2021-08-22 10:15:40.556] [info] Initializing Recorder (recorder)
[2021-08-22 10:15:40.557] [info] Initializing Rigctl Server (rigctl_server)
[2021-08-22 10:15:40.565] [info] Initializing SoapySDR Source (soapy_source)
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
[2021-08-22 10:15:41.532] [info] Setting sample rate to 250000.0
[2021-08-22 10:15:41.532] [info] New DSP samplerate: 250000.0 (source samplerate is 250000.0)
[2021-08-22 10:15:41.533] [info] Updating FFT DSP settings: Keep: 12500, Skip: 0
[2021-08-22 10:15:41.534] [info] Loading color maps
[2021-08-22 10:15:41.549] [info] Loading /usr/share/sdrpp/colormaps/gqrx.json
[2021-08-22 10:15:41.566] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2021-08-22 10:15:41.582] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2021-08-22 10:15:41.599] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2021-08-22 10:15:41.616] [info] Loading /usr/share/sdrpp/colormaps/classic_green.json
[2021-08-22 10:15:41.632] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2021-08-22 10:15:41.649] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2021-08-22 10:15:41.666] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2021-08-22 10:15:41.682] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2021-08-22 10:15:41.699] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2021-08-22 10:15:41.716] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2021-08-22 10:15:41.742] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2021-08-22 10:15:41.742] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2021-08-22 10:15:41.743] [info] RTLSDRSourceModule 'RTL-SDR Source': Menu Select!
[2021-08-22 10:15:41.743] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2021-08-22 10:15:41.743] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
terminate called after throwing an instance of 'RtAudioError'
  what():  RtApiPulse::stopStream(): the stream is not open!
Aborted
```

Ubuntu 18.04 is soooo close.

AlexandreRouma, 2021-08-24:
SDR++ now supports Ubuntu 18.04 without modification and a prebuilt deb is available both in the automated build and the futur releases.

If you want to build it yourself, you'll need an up to date cmake version (at least 3.13).

To use and build the airspyhf+ source, you'll need to install a recent version too