AlexandreRouma/SDRPlusPlus #264: Spyserver issue & suggestion

pbij2001, 2021-08-17:
When the spyserver is locked in frequency range, the sdrpp VFO still tunes outside the spyserver restricted range, losing correct frequency correlation.

-suggestion: make the spyserver IP field compatible with the spyserver URL format.
sdr://IP:port (all in one field)
This way it is easier to cut/copy from the spyserver map website in a single action
There is a ready-to-go "clipboard copy" on the website available.

AlexandreRouma, 2021-08-17:
That is a known issue and is the case for all SDR drivers, SDR++ doesn't currently have a way of limiting the frequencies the user can tune to.

As for the URL, I'll have to see if I can just add an automated decoding of the "spyserver" format, I personnaly really don't like that formatting so I think allowing to paste it into the host field and be automatically decoded would be a better option

pbij2001, 2021-08-17:
Manual entry is better in general.

Copy and decode would just be handy for the Spyserver net.
[image]

AlexandreRouma, 2022-09-13:
Won't implement in the end. I wouldn't recommend using the spyserver source anyway due to the limitations of the "open source" spyserver client.

I recommend switching to SDR++ Server.

closed 2022-09-13 by AlexandreRouma