AlexandreRouma/SDRPlusPlus #266: Periodic sound disortions

aso824, 2021-08-17:
I have some kind of periodic disortions, but I cannot figure out what is wrong. 
My configuration is Linux Mint 20 with RTL-SDR.
Everything works fine under gqrx.

On SDR++:
- disortions happens at regular intervals, approximately 10 seconds, and for duration of 2 to 3 seconds
- sounds like lowering the frequency of the sound in combination with chopping
- mute/squelch does not affect it - it's possible to unmute just at the moment with disortions
- any option like modulation etc. does not affect it; also, sound is fine on recording
- changing audio output or output rate "resets" the problem (probably reinitialisation of sink?), so first disortion will occur after ~10 seconds
- it's same at 11050Hz and on 96000Hz output, using my USB DAC or integrated soundcard - nothing changes
- cannot be recorded using built-in audio recorded (audio is perfectly fine)

I tried to record it with Audacity, but refused to start recording. Noticed that no other sounds from system is heard when audio in SDR++ is on. When I select `default` output instead of `hw` one, system sounds are heard along with SDR one, but I got bunch of errors and total disortion after few seconds:

`RtApiAlsa::callbackEvent: audio write error, underrun.`

Listen to attached files - first one (WFM, local radio) has disortions at 00:04, 00:14 and 00:24. Second one is noise with USB modulation turned on - it should be constant, but also defects are audible at regular intervals.

[sdrpp.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7002455/sdrpp.zip)

AlexandreRouma, 2021-08-17:
Hello,
This is a known issue with RtAudio, 

The solution is to compile RtAudio from source and make sure to force PulseAudio instead of direct Alsa.
The procedure is described by a user here: https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

aso824, 2021-08-18:
Thanks, it indeed helped. I suggest to add link to #112 with short explaination.

closed 2021-08-18 by aso824