AlexandreRouma/SDRPlusPlus #270: Windows version won't run?

LunaEros, 2021-08-18:
I unzipped the windows version and tried running it but it gives a dialog saying it can't start because SoapySDR.dll is missing.
How do I fix it?

AlexandreRouma, 2021-08-18:
What version of windows are you running?

LunaEros, 2021-08-18:
Win 7 SP1 x64

LunaEros, 2021-08-18:
Ok, I installed the latest Pothos SDR and it's soapysdrutil found my RTL-SDR v3 on device 4. So I went and tried SDR++ again and it didn't give me that error dialog but it ended up crashing after a few seconds before it completely loaded.

AlexandreRouma, 2021-08-18:
You don't need pothos if you're not gonna be using soapy SDR. Just delete modules/soapy_source.dll
Also Windows 7 SP1 is super old so you should probably update, I'm not sure how long I'll be able to keep supporting such ancient OSes...

LunaEros, 2021-08-18:
If I don't need it then why did it complain about not having soapy dlls and then crash?

AlexandreRouma, 2021-08-18:
Because windows 7 is stupid and instead of sending an error back to the code like all other operating systems, it crashes the entire software...

LunaEros, 2021-08-18:
So if I find this soapy_source.dll and delete it then SDR++ should work?

I'll probably eventually update to win10 I just hate new interfaces once I get a system to how I like it and also don't like what I've read about Microsoft spying on everything as well as installing updates when it feels like it. Which from what I've seen at one of my past jobs is all the time and at the least convenient moments.

AlexandreRouma, 2021-08-18:
It should work yes

LunaEros, 2021-08-18:
I found two instances and renamed them soapy_source.olddll but nope.  SDR++ got to the same spot.  It said initializing BladeRF source and then crashed.

AlexandreRouma, 2021-08-18:
could you send full startup logs by running SDR++ from a terminal with the `-s` parameter and copying everything it prints?

LunaEros, 2021-08-18:
C:\Video Edit\Radio\SDR++ x64>sdrpp.exe -s
[2021-08-18 13:55:44.405] [info] SDR++ v1.0.3
[2021-08-18 13:55:44.406] [info] Loading config
[2021-08-18 13:55:44.853] [info] Loading icons
[2021-08-18 13:55:44.873] [info] Loading band plans
[2021-08-18 13:55:44.881] [info] Loading band plans color table
[2021-08-18 13:55:44.886] [info] Loading modules
[2021-08-18 13:55:44.886] [info] Loading ./modules/airspyhf_source.dll
[2021-08-18 13:55:44.888] [info] Loading ./modules/airspy_source.dll
[2021-08-18 13:55:44.890] [info] Loading ./modules/audio_sink.dll
[2021-08-18 13:55:44.892] [info] Loading ./modules/bladerf_source.dll
[2021-08-18 13:55:44.895] [info] Loading ./modules/discord_integration.dll
[2021-08-18 13:55:44.896] [info] Loading ./modules/file_source.dll
[2021-08-18 13:55:44.898] [info] Loading ./modules/frequency_manager.dll
[2021-08-18 13:55:44.900] [info] Loading ./modules/hackrf_source.dll
[2021-08-18 13:55:44.902] [info] Loading ./modules/limesdr_source.dll
[2021-08-18 13:55:44.904] [info] Loading ./modules/meteor_demodulator.dll
[2021-08-18 13:55:44.906] [info] Loading ./modules/network_sink.dll
[2021-08-18 13:55:44.907] [info] Loading ./modules/plutosdr_source.dll
[2021-08-18 13:55:44.910] [info] Loading ./modules/radio.dll
[2021-08-18 13:55:44.911] [info] Loading ./modules/recorder.dll
[2021-08-18 13:55:44.913] [info] Loading ./modules/rigctl_server.dll
[2021-08-18 13:55:44.914] [info] Loading ./modules/rtl_sdr_source.dll
[2021-08-18 13:55:44.916] [info] Loading ./modules/rtl_tcp_source.dll
[2021-08-18 13:55:44.918] [info] Loading ./modules/sdrplay_source.dll
[2021-08-18 13:55:44.921] [info] Loading ./modules/spyserver_source.dll
[2021-08-18 13:55:44.923] [info] Initializing Airspy Source (airspy_source)
[2021-08-18 13:55:44.976] [info] Initializing AirspyHF+ Source (airspyhf_source)

[2021-08-18 13:55:45.026] [info] Initializing Audio Sink (audio_sink)
[2021-08-18 13:55:45.027] [info] Initializing BladeRF Source (bladerf_source)

AlexandreRouma, 2021-08-18:
try deleting modules/bladerf_source.dll.
A lot of those libraries aren't really meant to run on old systems I guess.
I know other people on windows 7 that run it just fine, I assume they have a later service pack or something...

LunaEros, 2021-08-18:
Ok. renamed that .oldll and tried again. Did the exact same thing. Even the -s output is identical.

LunaEros, 2021-08-18:
Don't pay attention to the last comment. I messed up changing it. I also had to do the same with HackRF_source.dll too since it got to that one and locked up.

LunaEros, 2021-08-18:
Ok. Either there's a real steep learning curve or something else is wrong. The program starts up now and I selected the RTL-SDR but when I click play nothing seems to happen. No sound (even though it's set on my speakers), no Spectrum, no waterfall. Nothing seems to be running. The only way it even looks like it's running is the source greys out when I hit play.

LunaEros, 2021-08-18:
Should I do the -s on the command line again?

AlexandreRouma, 2021-08-18:
> when I click play nothing seems to happen. 
Make sure you've installed the RTL-SDR drivers with Zadig like for all SDR software on windows. Instuctions are on rtl-sdr.com

LunaEros, 2021-08-19:
I have already.  It still works fine in SD Sharp.

LunaEros, 2021-08-21:
I tried running it again and changing various options with the -s running (output below).
The RTL-SDR works fine in SDR Sharp and HDRSDR.

C:\Video Edit\Radio\SDR++ x64>sdrpp.exe -s
[2021-08-21 14:15:29.827] [info] SDR++ v1.0.3
[2021-08-21 14:15:29.852] [info] Loading config
[2021-08-21 14:15:31.513] [info] Loading icons
[2021-08-21 14:15:31.575] [info] Loading band plans
[2021-08-21 14:15:31.639] [info] Loading band plans color table
[2021-08-21 14:15:31.805] [info] Loading modules
[2021-08-21 14:15:31.806] [info] Loading ./modules/airspyhf_source.dll
[2021-08-21 14:15:32.209] [info] Loading ./modules/airspy_source.dll
[2021-08-21 14:15:32.361] [info] Loading ./modules/audio_sink.dll
[2021-08-21 14:15:32.547] [info] Loading ./modules/discord_integration.dll
[2021-08-21 14:15:32.630] [info] Loading ./modules/file_source.dll
[2021-08-21 14:15:32.747] [info] Loading ./modules/frequency_manager.dll
[2021-08-21 14:15:32.830] [warning] ConfigManager locked, waiting...
[2021-08-21 14:15:32.841] [info] Loading ./modules/limesdr_source.dll
[2021-08-21 14:15:33.057] [info] Loading ./modules/meteor_demodulator.dll
[2021-08-21 14:15:33.180] [info] Loading ./modules/network_sink.dll
[2021-08-21 14:15:33.264] [info] Loading ./modules/plutosdr_source.dll
[2021-08-21 14:15:33.681] [info] Loading ./modules/radio.dll
[2021-08-21 14:15:33.822] [info] Loading ./modules/recorder.dll
[2021-08-21 14:15:33.934] [info] Loading ./modules/rigctl_server.dll
[2021-08-21 14:15:34.075] [info] Loading ./modules/rtl_sdr_source.dll
[2021-08-21 14:15:34.284] [info] Loading ./modules/rtl_tcp_source.dll
[2021-08-21 14:15:34.426] [info] Loading ./modules/sdrplay_source.dll
[2021-08-21 14:15:34.700] [info] Loading ./modules/spyserver_source.dll
[2021-08-21 14:15:34.824] [info] Initializing Airspy Source (airspy_source)
[2021-08-21 14:15:34.957] [info] Initializing AirspyHF+ Source (airspyhf_source)

[2021-08-21 14:15:35.036] [info] Initializing Audio Sink (audio_sink)
[2021-08-21 14:15:35.037] [info] Initializing BladeRF Source (bladerf_source)
[2021-08-21 14:15:35.037] [error] Module 'bladerf_source' doesn't exist
[2021-08-21 14:15:35.038] [info] Initializing File Source (file_source)
[2021-08-21 14:15:35.051] [info] Initializing Frequency Manager (frequency_manager)
[2021-08-21 14:15:35.079] [info] Initializing HackRF Source (hackrf_source)
[2021-08-21 14:15:35.080] [error] Module 'hackrf_source' doesn't exist
[2021-08-21 14:15:35.080] [info] Initializing LimeSDR Source (limesdr_source)
[2021-08-21 14:15:35.105] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-08-21 14:15:35.106] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-08-21 14:15:35.166] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-08-21 14:15:35.167] [info] Initializing Radio (radio)
[2021-08-21 14:15:35.329] [info] RtAudio stream open
[2021-08-21 14:15:35.365] [info] Initializing Recorder (recorder)
[2021-08-21 14:15:35.427] [info] Initializing Rigctl Server (rigctl_server)
[2021-08-21 14:15:35.429] [info] Initializing SDRplay Source (sdrplay_source)
[2021-08-21 14:15:35.517] [info] Initializing SoapySDR Source (soapy_source)
[2021-08-21 14:15:35.518] [error] Module 'soapy_source' doesn't exist
[2021-08-21 14:15:35.519] [info] Initializing SpyServer Source (spyserver_source)
[2021-08-21 14:15:35.521] [info] Loading color maps
[2021-08-21 14:15:35.522] [info] Loading ./res/colormaps/classic.json
[2021-08-21 14:15:35.525] [info] Loading ./res/colormaps/electric.json
[2021-08-21 14:15:35.528] [info] Loading ./res/colormaps/gqrx.json
[2021-08-21 14:15:35.558] [info] Loading ./res/colormaps/greyscale.json
[2021-08-21 14:15:35.560] [info] Loading ./res/colormaps/inferno.json
[2021-08-21 14:15:35.579] [info] Loading ./res/colormaps/magma.json
[2021-08-21 14:15:35.582] [info] Loading ./res/colormaps/plasma.json
[2021-08-21 14:15:35.589] [info] Loading ./res/colormaps/turbo.json
[2021-08-21 14:15:35.595] [info] Loading ./res/colormaps/viridis.json
[2021-08-21 14:15:35.598] [info] Loading ./res/colormaps/websdr.json
[2021-08-21 14:15:35.621] [info] New DSP samplerate: 250000.0 (source samplerate is 250000.0)
[2021-08-21 14:15:35.621] [info] Updating FFT DSP settings: Keep: 12500, Skip: 0

[2021-08-21 14:15:35.649] [info] RTLSDRSourceModule 'RTL-SDR Source': Menu Select!
[2021-08-21 14:15:35.651] [info] New DSP samplerate: 250000.0 (source samplerate is 250000.0)
[2021-08-21 14:15:35.652] [info] Updating FFT DSP settings: Keep: 12500, Skip: 0

[2021-08-21 14:15:35.696] [info] RtAudio stream open
[2021-08-21 14:15:35.717] [info] Updating FFT DSP settings: Keep: 12500, Skip: 0
[... 9 lines cut ...]
[2021-08-21 14:15:35.723] [info] Running post-init for File Source
[2021-08-21 14:15:35.723] [info] Running post-init for Frequency Manager
[2021-08-21 14:15:35.724] [info] Running post-init for LimeSDR Source
[2021-08-21 14:15:35.745] [info] Running post-init for PlutoSDR Source
[2021-08-21 14:15:35.745] [info] Running post-init for RTL-SDR Source
[2021-08-21 14:15:35.746] [info] Running post-init for RTL-TCP Source
[2021-08-21 14:15:35.746] [info] Running post-init for Radio
[2021-08-21 14:15:35.747] [info] Running post-init for Recorder
[2021-08-21 14:15:35.748] [info] Running post-init for Rigctl Server
[2021-08-21 14:15:35.748] [info] Running post-init for SDRplay Source
[2021-08-21 14:15:35.749] [info] Running post-init for SpyServer Source
[2021-08-21 14:15:35.749] [info] Ready.
[2021-08-21 14:15:56.639] [error] No device selected
[... cut ...]
[2021-08-21 14:19:10.132] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 103695200.0!
[2021-08-21 14:19:20.376] [error] No device selected
[2021-08-21 14:19:20.377] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 103695200.0!
[2021-08-21 14:19:40.763] [error] Will this make the software crash?
[2021-08-21 14:19:55.508] [info] New DSP samplerate: 2048000.0 (source samplerate is 2048000.0)
[2021-08-21 14:19:55.510] [info] Updating FFT DSP settings: Keep: 65536, Skip: 36864

C:\Video Edit\Radio\SDR++ x64>

AlexandreRouma, 2021-08-21:
It looks like it didn't find your RTL. Make sure no other software that might use it is running in the background.

SDR# keeps the device in use even if it's not selected, same with HDSDR, so make sure you have both closed

LunaEros, 2021-08-21:
I have. I thought of that.   When I was trying originally to put that Pothos soapysdr dlls in before you said I don't need that and uninstalled it the soapysdrutil.exe found it as device 4.
It's listed in my device manager as RTL2838UHIDIR under Universal Serial Bus Devices where Zadig assigned it.

aosync, 2022-08-17:
@LunaEros does the issue still happen?