AlexandreRouma/SDRPlusPlus #271: Add plugins from rtl-sdr.com?

EricFROL, 2021-08-19:
Is there a way to add plugins to SDRPP? It would be perfect and amazing.

shuyuan-liu, 2021-08-19:
Plugins for SDR# are currently not supported by SDR++. See #242.

EricFROL, 2021-08-19:
Any way to track a satélite using gpredict or orbitron?

> Plugins for SDR# are currently not supported by SDR++. See #242.

shuyuan-liu, 2021-08-19:
> Any way to track a satélite using gpredict or orbitron?

Yes, SDR++ has the rigctl_server module that accepts control from Gpredict. Load rigctl_server in the Module Manager tab (you'll need to give a name to the new tab that rigctl_server will use), open the newly created tab and click on Start. Then you can connect from Gpredict the usual way.

AlexandreRouma, 2021-08-19:
> Load rigctl_server in the Module Manager tab

I'll add that this shouldn't be needed if it's a new install

EricFROL, 2021-08-19:
I cant set a 37000 bandwith...

AlexandreRouma, 2021-08-19:
I assume you're likely trying to use WFM to decode APT; there are two things:

1) For  NOAA APT, use NFM, this will allow you to select 37k
2) I see you were asking about doppler tracking: NEVER use it for APT. APT is FM, it was desgined to not need doppler correction.
People who write tutorials saying to use it don't know what they're doing. It actually risks making the decode worse in some cases...

EricFROL, 2021-08-19:
> I assume you're likely trying to use WFM to decode APT; there are two things:
>
> 1. For  NOAA APT, use NFM, this will allow you to select 37k
> 2. I see you were asking about doppler tracking: NEVER use [...]

Well, I dont tottally agree with you... I can try to use NFM instead of WFM, but doppler effect exists... NOAA frequency is moving by the pass of the time.

AlexandreRouma, 2021-08-19:
NOAA uses FM with an AM subcarrier, this has the effect of cancelling the doppler shift automatically at decode when the AM subcarrier is demodulated. There is no reason do doppler correction. Yes, it moves, but if you use a bandwidth of 40KHz, it'll stay within the VFO's bandwidth and you'll recover the signal just fine.

You can still use doppler tracking *if you really want to* but except no difference (or even worse performance) for the extra effort of setting it up.