AlexandreRouma/SDRPlusPlus #274: Failed to initialize OpenGL loader

fjl82, 2021-08-19:
I just found out about SDR++ and I'm looking forward to try it. Except that it does not work. I get the below message about OpenGL. I saw another issue that turned out to be that person's local problem, but I don't think this is the same thing. I can use other OpenGL software, like UT2004 and Quake 4, just fine, so I don't think my OpenGL rendering has any issues.

```
$ sdrpp
[2021-08-19 15:44:43.047] [info] SDR++ v1.0.3
[2021-08-19 15:44:43.047] [info] Loading config
[2021-08-19 15:44:43.406] [error] Failed to initialize OpenGL loader!
```

I see a black window flashing by for a fraction of a second before it shuts down.

OpenGL information:

```
$ glxinfo|grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce 9500 GT/PCIe/SSE2
OpenGL core profile version string: 3.3.0 NVIDIA 340.108
OpenGL core profile shading language version string: 3.30 NVIDIA via Cg compiler
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.3.0 NVIDIA 340.108
OpenGL shading language version string: 3.30 NVIDIA via Cg compiler
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 NVIDIA 340.108 340.108
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.00
OpenGL ES profile extensions:
```

I'm using an Nvidia Geforce 9500 GT, with the 340.108 driver. Yes, I know it's an old card, and no, i'm not going to replace it, since I don't really play games on my pc (I have a playstation for that). You could probably see that from my retro game examples above ;) The 9500 suits my needs just fine.

I compiled SDR++ from source, since there is no rpm package for my distro (Fedora 34). I've installed glew 2.1.0 and glfw 3.3.4 and their development headers from the repository using dnf, and after disabling some plugins that I don't need (like airspy and plutosdr), it built fine as far as I can tell.

So my questions:
- Can you make SDR++ at least support software rendering, for people on lesser hardware or where it just refuses to work? I don't really understand anyway why an SDR software would need 3D at all.
- Is there any way to troubleshoot this? The information on stdout is not helpful. Does SDR++ have any commandline options to show debug information? -d, -v, -h and -? all don't work.

AlexandreRouma, 2021-08-19:
> Can you make SDR++ at least support software rendering, for people on lesser hardware or where it just refuses to work? I don't really understand anyway why an SDR software would need 3D at all.

Softare rendering can be achieved with a software rending OpenGL driver (I think someone here posted a link about it but I don't know much myself). OpenGL **isn't** just for 3D, it allows efficient hardware accelerated rendering of the UI, waterfall and FFT. I'm currently working on supporting OpenGL 2.0 too, but 3.0 should already be enough for 99% of PCs built in the last 10 years.

> s there any way to troubleshoot this? The information on stdout is not helpful. Does SDR++ have any commandline options to show debug information? -d, -v, -h and -? all don't work.

Unfortunately, glew doesn't give much more info when it fails to intialise, so there isn't much else I can display.

I know someone else has had the same issue and I still haven't managed to figure out what the issue is. Your card is recent enough to run SDR++, so the issue is something else.

fjl82, 2021-08-20:
So how can I get this to run? I don't want to replace my nvidia driver with something else with worse performance...

AlexandreRouma, 2021-08-20:
I think this might be related to #258 on OpenSuse.
Some distros seem to enable EGL which cuases issues with glew, i'll see if I can find a work around for this.

closed 2022-08-17 by AlexandreRouma