AlexandreRouma/SDRPlusPlus #275: Unicode support in bookmark names

shuyuan-liu, 2021-08-19:
Currently (as of 9f0e050d1b9377d32a80c2b3750c64d902fd7c3f) both Cyrillic and CJK (Chinese, Japanese, Korean) characters in bookmark names are displayed as question marks in Frequency Manager. This is especially inconvenient when creating bookmarks for e.g. broadcast FM stations that have their names only in a language not using the Roman alphabet. I suspect this is also the case in all other parts of the UI but I didn't test it.

[image]

frequency_manager_config.json (frequencies were arbitrarily picked):
```
{
    "bookmarkDisplayMode": 1,
    "lists": {
        "Broadcast FM in Guangzhou": {
            "bookmarks": {
                "русский язык": {
                    "bandwidth": 200000.0,
                    "frequency": 200000000.0,
                    "mode": 1
                },
                "日本語": {
                    "bandwidth": 200000.0,
                    "frequency": 201000000.0,
                    "mode": 1
                },
                "简体中文": {
                    "bandwidth": 200000.0,
                    "frequency": 200500000.0,
                    "mode": 1
                },
                "한국어/조선말": {
                    "bandwidth": 200000.0,
                    "frequency": 201500000.0,
                    "mode": 1
                }
            },
            "showOnWaterfall": true
        }
    },
    "selectedList": "General"
}
```

AlexandreRouma, 2021-08-19:
Hello,
Support for these characters isn't planed in the near future. Although ImGui is supposed to support UTF8, I've never gotten it to work with non latin characters.

shuyuan-liu, 2021-08-19:
That's alright, guess I'll stick to SDR# for FM radio for the time being. Closing the issue for now.

closed 2021-08-19 by shuyuan-liu