AlexandreRouma/SDRPlusPlus #277: Abnormal CPU usage

eloysl, 2021-08-19:
Disclaimer: I am not used to asking questions on Github about this kind of problems, so if this is not the correct way or place to do it, please let me know how or where I should ask this.

I am running SDR++ on Pop OS 20.04 (closely based on Ubuntu) on a laptop with a 4-core Intel i7-4510U CPU at 2 GHz, and I have detected that the CPU usage rises to the maximum (between 85% and 95% on all 4 cores) when I open the program and it goes back to normal (around 15%) when I close it. This happens consistently every time I run SDR++.

To be more specific, when this happens, the process ```sdrpp``` accounts for around 50% of CPU usage, and ```gnome-shell``` takes between 30% and 40%. At least in the latter case, these values don't seem very normal. What is curious is that all this happens without playing the device, just by opening the program.

I don't know how to provide more info that could help determine what is happening, but I'll be happy to follow "for dummies" indications in order to do so.

AlexandreRouma, 2021-08-19:
Could you check the debug tab in the menu and tell me at what framerate SDR++ is running?
(or just take a screenshot of that menu)

Also a screenshot of htop with and without SDR++ open would be useful

eloysl, 2021-08-20:
Sure!

According to the debug tab, SDR++ is running at around 60 fps, ocasionally dropping to 45-48 and then going back up. I attach a screenshot of that menu and also screenshots of htop with and without SDR++

[image]
[image]
[image]

AlexandreRouma, 2021-08-21:
Alright that's rather weird, Are you by any chance running in software rendering mode for some reason? (eg. Mesa3D)

eloysl, 2021-08-21:
Probably yes, I guess :/ It turned out that I had a graphics driver missing. After installing it, everything works fine. Sorry for the inconvenience, it was a fault on my side.