AlexandreRouma/SDRPlusPlus #278: HRPT Reception/Bug

zeke800, 2021-08-19:
Hello,

I wanted to receive HRPT from NOAA sats, and I am getting a strong signal at 1698 and 1707 MHz. Unfortunately, I noticed a couple bugs that prevent me from recording the signal. Please don't get me wrong, this is a wonderful piece of software and I am so glad that Alexandre Rouma decided to write it. I am using Windows 10. 
Bug 1:
When I tune to 1698 MHz (NOAA 19 frequency), I can sometimes see it as off to the side, so I can't see the full HRPT signal. 
Question 2:
How do I compile weather_sat_decoder in this repository? I want to decode images like [this](https://pbs.twimg.com/media/EyfAaagWgAEnonI?format=jpg&name=4096x4096).

AlexandreRouma, 2021-08-19:
Hello,

> When I tune to 1698 MHz (NOAA 19 frequency), I can sometimes see it as off to the side, so I can't see the full HRPT signal

This is normal, the satellite has doppler shift, you need to either correct it using a tracker like gpredict connected through the rigctl server (NOT recommended), or use an SDR with a wide enough to accomodate the shift.
However, if, you use a RTL-SDR (not really recommended for HRPT anyway), there is an issue is a known issue where the RTL-SDR doesn't tune to the right to the frequency when instructed to. To test, simply start and stop the RTL-SDR after tuning to the right place. (though this generally is mostly an issue on linux)

> How do I compile weather_sat_decoder in this repository? I want to decode images like this. Could somebody please provide precompiled DLLs?

The weather sat decoder, as mentioned in the readme, is unfinished and not ready for use. I recommend using a third party decoder like SatDump until this module is finished which might take a few months.

zeke800, 2021-08-19:
Thank you so much for your fast reply!! Sorry, for the shift if frequency I don't mean doppler shift. When I turn on SDR++, sometimes when I tune to 1698 it isn't in the center of the graph. Very weird, so I thought I would report it. Also, can I please beta-test the DLL extension? It is really useful because you don't need to make 4Gb recordings through satdump.

zeke800, 2021-08-19:
I would be extremely grateful if you could do so :)

AlexandreRouma, 2021-08-19:
> Thank you so much for your fast reply!! Sorry, for the shift if frequency I don't mean doppler shift. When I turn on SDR++, sometimes when I tune to 1698 it isn't in the center of the graph

Ah no, I see what you mean, make sure you've got center tuning enabled (the button next to the frequency select) when it shows a crosshair, the selected frequency will the kept in the middle of the FFT.

> can I please beta-test the DLL extension?

By unifinished I mean it's unusable. It doesn't save anything, it just decodes live and is still missing decoding for other instruments of the satellite, as well as post processing of the images. I need to get back to working on this but it'll take more time to have it actually be useful.

zeke800, 2021-08-20:
Ok, maybe I'll try to help with the development. But thank you for all your support!!

zeke800, 2021-08-20:
Wait, but if I want to develop, how can I set it up on my computer?

zeke800, 2021-08-20:
Actually, never mind, I'll figure it out myself. Thank you for all the support, and sorry for the inconveniences!

closed 2021-08-20 by zeke800