AlexandreRouma/SDRPlusPlus #281: [Feature] Mouse wheel zoom

arkhnchul, 2021-08-20:
patch in attachment, it allows to change zoom level with mouse wheel while holding Shift button.

short demonstration: https://youtu.be/hoT2pMBhOsA

[0001-mouse-wheel-zoom-implementation.patch.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7020904/0001-mouse-wheel-zoom-implementation.patch.txt)

AlexandreRouma, 2021-08-20:
Hello,

Thanks for submitting this patch but unfortunately I think this is quite a dirty way of implementing this feature, I'll see if I can add this to my todo list.

arkhnchul, 2021-08-21:
maybe i can remade it in more appropriate way?

AlexandreRouma, 2022-09-30:
After reflection I won't be adding that feature. I scrolling (including with modifiers) will be reserved for other things.

closed 2022-09-30 by AlexandreRouma