AlexandreRouma/SDRPlusPlus #286: AppImage version of SDRPlusPlus would be very useful

N0rbert, 2021-08-20:
Please prepare and AppImage version of SDRPlusPlus would be very useful.
It will be able to run on Ubuntu 18.04 LTS which is still actual and alive.

AlexandreRouma, 2021-08-20:
Appimages are bad for this type of software.
I'm already working on supporting older distros, but Ubuntu 18.04 is so out of date that you'll be losing on performance.

N0rbert, 2021-08-20:
I should note that CubicSDR is distributed [also as AppImage](https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4).

Currently I need to run additional Ubuntu 20.04 LTS VM to test the application.

AlexandreRouma, 2021-08-20:
CubicSDR is nowhere near as complex as SDR++
You can't just put it in an appimage and hope it works.
it has to be installed properly for all modules to be loaded

I'll add that running it in a VM won't work unless you enable software rendering for opengl

N0rbert, 2021-08-20:
> I'll add that running it in a VM won't work unless you enable software rendering for opengl

So this is why I kindly as for official package for Ubuntu 18.04 LTS.

AlexandreRouma, 2021-08-20:
As I said, Ubuntu 18.04 is so old that it only supports volk 1.3 AND doesn't support C++17 features like std::filesystem
These are not things I can support easily without changing thousands of lines of code, aka, I won't be able to do it quickly.

leirbag28, 2025-02-08:
I want to Re-Open this Request. I think it is Now VITAL that SDRplusplus come as an Appimage. It just makes sence as a simple drag and Drop on any Linux Distro to run. All dependencies packed right in there without affecting the System/OS. We even Have Microsoft Windows Apps packaged as Appimage/GameImage/FLatImage now. It just works! even on Ubuntu 18.04 LTS