AlexandreRouma/SDRPlusPlus #287: Installation on Ubuntu 20.04 LTS requires installation libopengl0 which is not listed dependency inside deb-package

N0rbert, 2021-08-20:
I have fully updated Ubuntu MATE 20.04 LTS without PPAs and thirdparty repositories.

Here I have downloaded the official deb-package from releases page as follows:

```
cd ~/Downloads
wget -c https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.3/sdrpp_ubuntu_focal_amd64.deb
sudo apt-get install ./sdrpp_ubuntu_focal_amd64.deb
```

it installed with the following messages:

```
$ sudo apt install ./sdrpp_ubuntu_focal_amd64.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'sdrpp' instead of './sdrpp_ubuntu_focal_amd64.deb'
The following NEW packages will be installed:
  sdrpp
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/2 532 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/mate/Downloads/sdrpp_ubuntu_focal_amd64.deb sdrpp all 0.2.5-282 [2 532 kB]
Selecting previously unselected package sdrpp.
(Reading database ... 291656 files and directories currently installed.)
Preparing to unpack .../sdrpp_ubuntu_focal_amd64.deb ...
Unpacking sdrpp (0.2.5-282) ...
Setting up sdrpp (0.2.5-282) ...
Processing triggers for desktop-file-utils (0.24-1ubuntu3) ...
Processing triggers for mime-support (3.64ubuntu1) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu1) ...
Processing triggers for bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu2) ...
Rebuilding /usr/share/applications/bamf-2.index...
```

on first launch it complained about missing library

```
$ sdrpp 
sdrpp: error while loading shared libraries: libOpenGL.so.0: cannot open shared object file: No such file or directory
```

so I need to [find a package](https://packages.ubuntu.com/search?suite=focal&arch=amd64&mode=exactfilename&searchon=contents&keywords=libOpenGL.so.0) for this file, I installed it:

```
sudo apt install libopengl0
```

and then `sdrpp` launched.

Please adjust package dependencies - add `libopengl0` package to Depends field.

N0rbert, 2021-08-20:
Moreover `sdrpp` package has empty depends session currently:

```
$ apt-cache depends sdrpp 
sdrpp
```

Please carefully follow https://www.debian.org/doc/manuals/maint-guide/ .

AlexandreRouma, 2021-08-20:
Follow the readme on what depdencies have to be installed. installing libopengl makes no sense, your graphics driver is in charge of this.

The dependency field is left empty on purpose as because every users will only install what they need according to their hardware