AlexandreRouma/SDRPlusPlus #299: [FEATURE REQUEST] Frequency Scanner.

grappas, 2021-08-22:
It would be nice if there was some of that feature - at least based on bookmarks if not scanning whole bandwidth.

JosephTheTank13, 2021-08-22:
I second this!

AlexandreRouma, 2021-08-22:
A frequency scanner feature is already in the works, you can actually already see the `scanner` folder in the source code

AlexandreRouma, 2021-08-22:
I'll update this thread once the feature is properly implemented

grappas, 2021-08-22:
It would be nice if it been implemented like original Frequency Scanner plugin in SDR#

jaymot123, 2021-08-28:
I came here to request this and it's already in the works. Great news.

perseus68-github, 2021-08-28:
Definitely a"must have" for any top-rated SDR software. Can't wait to see it implemented 👌

grappas, 2021-08-28:
@AlexandreRouma If I may - you said "once the feature is properly implemented" - even if It isn't (as you said) "properly" implemented but implemented nonetheless: Is there any way to activate it or test it. I'm using aur package and if it picks code straight from main repo, than it should be some way to test it right?

AlexandreRouma, 2021-08-28:
there's just the skeleton of the module but it doesn't do anything

Wnfidencio, 2021-12-15:
I think good changes to the SDR# frequency scanner would be to put a schedule, so the radio changes what it's scanning accordingly. Also clarify the threshold levels that the red and yellow lines represent, letting the user choose which SNR he wants to use in a box.
I believe that the scanner must have a list of canceled frequencies, and that this list can be downloaded and edited in other software, or in SDR++ itself, so that we can easily add and remove frequencies from this blacklist.
I believe that the scanner doesn't make sense without an audio recorder, which could also be improved and facilitated, especially in the matter of how to save file names. And also create a log of recorded audio where the user can launch other features of that signal and download everything in a .csv for example.

Wnfidencio, 2021-12-16:
I think that just one plugin could just concatenate the functions of Frequency Scanner, Frequency manager and Audio Recorder with the addition of a scheduler and more signal classification options.

BT-Justice, 2022-06-08:
> I'll update this thread once the feature is properly implemented

Hopefully soon.  Miss using my Airspy as a scanner.

icepicknz, 2022-07-12:
interested too; something to scan a group loaded from frequency manager

gbeyer3, 2022-07-12:
Hi, I'm not seeing a scanner folder in the source code per 8/222/21 post by Alexandre.  Am I missing something?  Thanks.

grappas, 2022-07-13:
I've bumped recently on that scanner module for gqrx. Made myself a simple modification (that yellow an red line from airspy ;))

https://github.com/neural75/gqrx-scanner/pull/20

That on linux it's not all lost ;)

AlexandreRouma, 2022-09-13:
Added and available in the latest nightly, development will continue.

closed 2022-09-13 by AlexandreRouma

grappas, 2022-09-14:
But there's no bookmark scanner 😒

grappas, 2022-12-14:
😥