AlexandreRouma/SDRPlusPlus #320: Minion SDR connect to SDR++

chrisfriro, 2021-08-28:
I tried to get a connection via USB between my minion sdr (qrpver.com) and a windows 10, but I couldn’t find a solution Does anybody has a solution or an idea? In a few days I will also try to get this couple run with a Debian based system. Same question will arise 👍 73! Christoph

AlexandreRouma, 2021-08-28:
If it's supported by soapysdr, you can use it with the SoapySDR source.

However I don't own one of those so I can't test or add support for it

chrisfriro, 2021-08-29:
Dear Alexandre,
I can not select SoapySDR source in the module manager. It is named in the **config**, but I can not see/find the soapySDR.dll after unzipping the *.windows.zip* in the registry.
73! Christoph

AlexandreRouma, 2021-08-29:
This is normal, you need to have soapysdr installed on your system for this.

You can install it by installing PothosSDR (make sure to check the add to path option in the installer)

closed 2022-09-13 by AlexandreRouma