AlexandreRouma/SDRPlusPlus #321: Segmentation fault (Ubuntu 20.0.4 and Lenovo T61 laptop)

9a7aof, 2021-08-28:
Hello,

I downloaded and compiled today SDRPlus v1.0.4 without problems.
When I started it, this appeared:

darko@lenovo:~/src/SDRPlusPlus/build$ sdrpp 
[2021-08-28 09:26:25.110] [info] SDR++ v1.0.4
[2021-08-28 09:26:25.110] [info] Loading config
[2021-08-28 09:26:25.111] [warning] ConfigManager locked, waiting...
[2021-08-28 09:26:25.461] [info] Using OpenGL 3.0
Segmentacijska greška (**krah**: stanje (dump) memorije je zapisano!)
darko@lenovo:~/src/SDRPlusPlus/build$ 

I don't know what the error is, on other computers compiling the startup program goes fine.
Thanks!

Darko, 9a7aof

AlexandreRouma, 2021-08-28:
Could you give more info on your PC?
CPU, RAM, `glxinfo | grep OpenGL` to get info about OpenGL, etc...

9a7aof, 2021-08-28:
This is info from my PC:

[cpuinfo.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070870/cpuinfo.tx
[OpenGL.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070873/OpenGL.txt)
t)
[memory.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070871/memory.txt)

9a7aof, 2021-08-28:
This is info for my PC:

[OpenGL.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070881/OpenGL.txt)
[memory.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070882/memory.txt)
[cpuinfo.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/7070883/cpuinfo.txt)

Thanks!

AlexandreRouma, 2021-08-28:
Alright, I think the issue here is you're running out of ram. You ownly have 373MB of ram available, this isn't even eneough to hold the waterfall history in some cases.

SDR++ Needs at least around 1GB free when running at 1080p

9a7aof, 2021-08-28:
Now I have 1.5GB ram free,.. Segmentation fault...
darko@lenovo:~$ free -h
              total        used        free      shared  buff/cache   available
Mem:          1,9Gi       285Mi       972Mi       2,0Mi       696Mi       1,5Gi
Swap:         2,0Gi          0B       2,0Gi
darko@lenovo:~$ sdrpp
[2021-08-28 18:12:21.969] [info] SDR++ v1.0.4
[2021-08-28 18:12:21.970] [info] Loading config
[2021-08-28 18:12:21.971] [warning] ConfigManager locked, waiting...
[2021-08-28 18:12:22.106] [info] Using OpenGL 3.0
Segmentacijska greška (**krah**: stanje (dump) memorije je zapisano!)
darko@lenovo:~$

AlexandreRouma, 2021-08-29:
Still pretty sure that it's running out of ram, that's basically the only thing that could cause that to happen...
That PC seems extremely old and under powered for SDR stuff

Sinager, 2021-09-22:
Same issue here on my Lenovo TP X230, Ubuntu 20.04LTS and SDRPP built from source
Ram 7,5 GiB
Intel® Core™ i5-3320M CPU @ 2.60GHz × 4 

              total        used        free      shared  buff/cache   available
Mem:          7,5Gi       1,6Gi       2,5Gi       362Mi       3,4Gi       5,3Gi
Swap:         2,0Gi        25Mi       2,0Gi

OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4000 (IVB GT2)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 21.0.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

sinager@SSTP:~/SDRPlusPlus/build$ sdrpp
[2021-09-22 10:32:48.892] [info] SDR++ v1.0.4
[2021-09-22 10:32:48.892] [info] Loading config
[2021-09-22 10:32:48.968] [info] Using OpenGL 3.0
[2021-09-22 10:32:49.114] [info] Loading icons
[2021-09-22 10:32:49.133] [info] Loading band plans
[2021-09-22 10:32:49.136] [info] Loading band plans color table
[2021-09-22 10:32:49.145] [info] Loading modules
[2021-09-22 10:32:49.145] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2021-09-22 10:32:49.152] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2021-09-22 10:32:49.169] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2021-09-22 10:32:49.186] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2021-09-22 10:32:49.202] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2021-09-22 10:32:49.225] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2021-09-22 10:32:49.235] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2021-09-22 10:32:49.252] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2021-09-22 10:32:49.268] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2021-09-22 10:32:49.286] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2021-09-22 10:32:49.302] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2021-09-22 10:32:49.319] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2021-09-22 10:32:49.335] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2021-09-22 10:32:49.354] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2021-09-22 10:32:49.368] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2021-09-22 10:32:49.385] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2021-09-22 10:32:49.402] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2021-09-22 10:32:49.418] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2021-09-22 10:32:49.436] [info] Initializing Airspy Source (airspy_source)
[2021-09-22 10:32:49.463] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-09-22 10:32:49.472] [info] Initializing Audio Sink (audio_sink)
[2021-09-22 10:32:49.484] [info] Initializing File Source (file_source)
[2021-09-22 10:32:49.501] [info] Initializing Frequency Manager (frequency_manager)
[2021-09-22 10:32:49.517] [info] Initializing HackRF Source (hackrf_source)
[2021-09-22 10:32:49.548] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-09-22 10:32:49.550] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-09-22 10:32:49.577] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-09-22 10:32:49.583] [info] Initializing Radio (radio)
Errore di segmentazione (core dump creato)

AlexandreRouma, 2021-09-22:
I'll have to go over the figure and figure out where the double free issue would possibly come from...

AlexandreRouma, 2021-09-24:
Alright, could anyone here who can get it to crash consistently help by running a special debug version?
I'd like to get a detailed backtrace of where the issue occurs.
I'll send all files needed.

Sinager, 2021-09-24:
Sure: give me directions.
Thanks
Stefano IZ0MJE

AlexandreRouma, 2021-09-24:
Alright, just install [this deb file](https://drive.google.com/file/d/1Qbg_MoWqiPRtZtBuFFdjroErbrscvTlf/view?usp=sharing) on top of your existing SDR++ installation.

Then, make sure gdb is installed and run `gdb /usr/bin/sdrpp`. When it prompts you to run a command, type `run` and press enter. It should start SDR++ and when it eventually crashes, it'll print a bunch of debug information to the console. Send back all of what it sent and I'll have a look.

Sinager, 2021-09-24:
Hello,
here we are.
Stefano

Reading symbols from /usr/bin/sdrpp...
(gdb) run
Starting program: /usr/bin/sdrpp
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2021-09-24 18:23:18.347] [info] SDR++ v1.0.4
[2021-09-24 18:23:18.348] [info] Loading config
[New Thread 0x7ffff07d8700 (LWP 5238)]
[New Thread 0x7fffee917700 (LWP 5239)]
[New Thread 0x7fffee116700 (LWP 5240)]
[New Thread 0x7fffed915700 (LWP 5241)]
[New Thread 0x7fffed114700 (LWP 5242)]
[2021-09-24 18:23:18.501] [info] Using OpenGL 3.0
[2021-09-24 18:23:18.880] [info] Loading icons
[2021-09-24 18:23:18.904] [info] Loading band plans
[2021-09-24 18:23:18.922] [info] Loading band plans color table
[New Thread 0x7fffc8bbf700 (LWP 5243)]
[New Thread 0x7fffc83be700 (LWP 5244)]
[New Thread 0x7fffc7bbd700 (LWP 5245)]
[New Thread 0x7fffc73bc700 (LWP 5246)]
[New Thread 0x7fffc6bbb700 (LWP 5247)]
[New Thread 0x7fffc63ba700 (LWP 5248)]
[2021-09-24 18:23:18.930] [info] Loading modules
[2021-09-24 18:23:18.930] [info] Loading
/usr/lib/sdrpp/plugins/meteor_demodulator.so
[New Thread 0x7fffc5bb9700 (LWP 5249)]
[2021-09-24 18:23:19.363] [info] Loading
/usr/lib/sdrpp/plugins/soapy_source.so
[New Thread 0x7fffc5355700 (LWP 5250)]
[2021-09-24 18:23:19.802] [info] Loading
/usr/lib/sdrpp/plugins/rtl_sdr_source.so
[New Thread 0x7fffbffff700 (LWP 5251)]
[2021-09-24 18:23:20.231] [info] Loading
/usr/lib/sdrpp/plugins/frequency_manager.so
[New Thread 0x7fffbf7fe700 (LWP 5252)]
[2021-09-24 18:23:20.812] [warning] ConfigManager locked, waiting...
[2021-09-24 18:23:20.813] [info] Loading
/usr/lib/sdrpp/plugins/audio_sink.so
[New Thread 0x7fffbe677700 (LWP 5253)]
[2021-09-24 18:23:21.263] [info] Loading
/usr/lib/sdrpp/plugins/network_sink.so
[New Thread 0x7fffbdbdb700 (LWP 5254)]
[2021-09-24 18:23:21.682] [info] Loading
/usr/lib/sdrpp/plugins/rigctl_server.so
[New Thread 0x7fffbd14b700 (LWP 5255)]
[2021-09-24 18:23:22.118] [info] Loading
/usr/lib/sdrpp/plugins/hackrf_source.so
[New Thread 0x7fffb3fff700 (LWP 5256)]
[2021-09-24 18:23:22.530] [info] Loading
/usr/lib/sdrpp/plugins/airspy_source.so
[New Thread 0x7fffb37fe700 (LWP 5257)]
[2021-09-24 18:23:22.937] [info] Loading
/usr/lib/sdrpp/plugins/discord_integration.so
[2021-09-24 18:23:22.951] [info] Loading
/usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2021-09-24 18:23:23.270] [info] Loading
/usr/lib/sdrpp/plugins/spyserver_source.so
[New Thread 0x7fffb2b0e700 (LWP 5258)]
[2021-09-24 18:23:23.692] [info] Loading
/usr/lib/sdrpp/plugins/plutosdr_source.so
[New Thread 0x7fffafe9c700 (LWP 5259)]
[2021-09-24 18:23:24.103] [info] Loading
/usr/lib/sdrpp/plugins/sdrplay_source.so
[New Thread 0x7fffaf206700 (LWP 5260)]
[2021-09-24 18:23:24.516] [info] Loading
/usr/lib/sdrpp/plugins/rtl_tcp_source.so
[New Thread 0x7fffae78e700 (LWP 5261)]
[2021-09-24 18:23:24.903] [info] Loading
/usr/lib/sdrpp/plugins/new_portaudio_sink.so
[New Thread 0x7fffadf8d700 (LWP 5262)]
[2021-09-24 18:23:24.915] [info] Loading
/usr/lib/sdrpp/plugins/limesdr_source.so
[2021-09-24 18:23:24.924] [error] Couldn't load
/usr/lib/sdrpp/plugins/limesdr_source.so.
[2021-09-24 18:23:24.924] [info] Loading
/usr/lib/sdrpp/plugins/file_source.so
[New Thread 0x7fffad478700 (LWP 5263)]
[2021-09-24 18:23:25.402] [info] Loading
/usr/lib/sdrpp/plugins/bladerf_source.so
[New Thread 0x7fffac9ef700 (LWP 5264)]
[2021-09-24 18:23:25.813] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[New Thread 0x7fffabe9b700 (LWP 5265)]
[2021-09-24 18:23:26.346] [info] Loading
/usr/lib/sdrpp/plugins/airspyhf_source.so
[New Thread 0x7fffab41e700 (LWP 5266)]
[2021-09-24 18:23:26.747] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[New Thread 0x7fffaa940700 (LWP 5267)]
[2021-09-24 18:23:27.177] [info] Initializing Airspy Source (airspy_source)
[New Thread 0x7fffa91fb700 (LWP 5268)]
[Thread 0x7fffa91fb700 (LWP 5268) exited]
[2021-09-24 18:23:27.185] [info] Initializing AirspyHF+ Source
(airspyhf_source)
[New Thread 0x7fffa91fb700 (LWP 5269)]
[Thread 0x7fffa91fb700 (LWP 5269) exited]
[2021-09-24 18:23:27.193] [info] Initializing Audio Sink (audio_sink)
[2021-09-24 18:23:27.194] [info] Initializing File Source (file_source)
[2021-09-24 18:23:27.196] [info] Initializing Frequency Manager
(frequency_manager)
[2021-09-24 18:23:27.205] [info] Initializing HackRF Source (hackrf_source)
[New Thread 0x7fffa91fb700 (LWP 5270)]
[2021-09-24 18:23:27.230] [info] Initializing PlutoSDR Source
(plutosdr_source)
[2021-09-24 18:23:27.239] [info] Initializing RTL-SDR Source
(rtl_sdr_source)
[2021-09-24 18:23:27.266] [info] Initializing RTL-TCP Source
(rtl_tcp_source)
[2021-09-24 18:23:27.271] [info] Initializing Radio (radio)
[New Thread 0x7fffc7bbd700 (LWP 5271)]
[Thread 0x7fffc7bbd700 (LWP 5245) exited]
[New Thread 0x7fff76178700 (LWP 5272)]
[New Thread 0x7fff75977700 (LWP 5273)]

Thread 1 "sdrpp" received signal SIGSEGV, Segmentation fault.
0x00007ffff7a2fd7f in
__gnu_cxx::new_allocator<dsp::untyped_stream*>::construct<dsp::untyped_stream*,
dsp::untyped_stream* const&> (this=0x5555562944e8, __p=0x0) at
/usr/include/c++/9/ext/new_allocator.h:147
147 { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
(gdb)

AlexandreRouma, 2021-09-24:
Alright, could you run `bt full` in GDB before running `run`? I thought by default it would print a bit more

Sinager, 2021-09-24:
Attached.

[2021-09-24 18:40:51.247] [info] Initializing Radio (radio)
[Thread 0x7fffc7bbd700 (LWP 6118) exited]
[New Thread 0x7fffc7bbd700 (LWP 6144)]
[New Thread 0x7fff76178700 (LWP 6145)]
[New Thread 0x7fff75977700 (LWP 6146)]
malloc(): unsorted double linked list corrupted

Thread 1 "sdrpp" received signal SIGABRT, Aborted.
__GI_raise ***@***.***=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: File o directory non esistente.
(gdb) bt full
#0  __GI_raise ***@***.***=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 10656, 8, 0, 0, 0, 0, 281470681751474, 0, 0, 0, 0, 0, 0, 0, 0}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007ffff70ae859 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x3feff621e3796d7e, sa_sigaction = 0x3feff621e3796d7e}, sa_mask = {__val = {0, 4586913746847464361, 0 <repeats 13 times>, 2}}, sa_flags = 0, sa_restorer = 0x0}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007ffff71193ee in __libc_message ***@***.***=do_abort, ***@***.***=0x7ffff7243285 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
        ap = {{gp_offset = 24, fp_offset = 0, overflow_arg_area = 0x7fffffffd140, reg_save_area = 0x7fffffffd0d0}}
        fd = <optimized out>
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#3  0x00007ffff712147c in malloc_printerr ***@***.***=0x7ffff7245ad8 "malloc(): unsorted double linked list corrupted") at malloc.c:5347
No locals.
#4  0x00007ffff712446c in _int_malloc ***@***.***=0x7ffff7274b80 <main_arena>, ***@***.***=16000080) at malloc.c:3744
        next = <optimized out>
        iters = <optimized out>
        nb = <optimized out>
        idx = 126
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = 0
        tcache_nb = 0
        tc_idx = 1000004
        return_cached = <optimized out>
        __PRETTY_FUNCTION__ = "_int_malloc"
#5  0x00007ffff7124c5f in _int_memalign ***@***.***=0x7ffff7274b80 <main_arena>, ***@***.***=32, ***@***.***=16000000) at malloc.c:4684
        nb = <optimized out>
        m = <optimized out>
        p = <optimized out>
        brk = <optimized out>
        newp = <optimized out>
        newsize = <optimized out>
        leadsize = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        size = <optimized out>
        __PRETTY_FUNCTION__ = "_int_memalign"
#6  0x00007ffff712751c in _mid_memalign (address=<optimized out>, bytes=16000000, alignment=32) at malloc.c:3312
        ar_ptr = 0x7ffff7274b80 <main_arena>
       p = <optimized out>
        hook = <optimized out>
        ar_ptr = <optimized out>
        p = <optimized out>
        hook = <optimized out>
[... 78 lines cut ...]
            height = 196, pixels = 0x555555a95260 "\001\374\377\235\004<"}, {width = 256, height = 256, 
            pixels = 0x555555abaab0 "\002\275\002\277\002\301\002\304\002\306\002\310\002\312\002\314\002\316\002\320\002\322\002\324\002\330\002\332\002\334\002\336\002\340\002\342\002\344\002\346\002\350\002\352\002\354\002\356\002\360\002\361\002\363\002\365\003R\003S\003T\003U\003V\003W\003X\003Z\003[\003\\\003]\003^\003_\003`\003a\003c\003d\003e\003f\003g\003h\003i\003y\003z\003{\003|\003}\003~\003\177\003\200\003\201\003\202\003\203\003\204\003\205\003\206\003\207\003\210\003\211\003\212\003\213\003\214\003\215\003\216\003\272\003\274\003\276\003\323\003\331\003\337\004H\004J\004N\004V\004X\004]\004i"}}
        io = @0x555555a8e6b8: {ConfigFlags = 0, BackendFlags = 6, DisplaySize = {x = 1280, y = 518}, DeltaTime = 0.0115804002, IniSavingRate = 5, IniFilename = 0x0, LogFilename = 0x7ffff7a989d2 "imgui_log.txt", 
          MouseDoubleClickTime = 0.300000012, MouseDoubleClickMaxDist = 6, MouseDragThreshold = 6, KeyMap = {258, 263, 262, 265, 264, 266, 267, 268, 269, 260, 261, 259, 32, 257, 256, 335, 65, 67, 86, 88, 89, 90}, 
          KeyRepeatDelay = 0.275000006, KeyRepeatRate = 0.0500000007, UserData = 0x0, Fonts = 0x555555a925a0, FontGlobalScale = 1, FontAllowUserScaling = false, FontDefault = 0x0, DisplayFramebufferScale = {x = 1, 
            y = 1}, MouseDrawCursor = false, ConfigMacOSXBehaviors = false, ConfigInputTextCursorBlink = true, ConfigDragClickToInputText = false, ConfigWindowsResizeFromEdges = true, 
          ConfigWindowsMoveFromTitleBarOnly = false, ConfigMemoryCompactTimer = 60, BackendPlatformName = 0x7ffff7ab3c40 "imgui_impl_glfw", BackendRendererName = 0x7ffff7ab3d7e "imgui_impl_opengl3", 
          BackendPlatformUserData = 0x0, BackendRendererUserData = 0x0, BackendLanguageUserData = 0x0, GetClipboardTextFn = 0x7ffff79d8fa7 <ImGui_ImplGlfw_GetClipboardText(void*)>, 
          SetClipboardTextFn = 0x7ffff79d8fc5 <ImGui_ImplGlfw_SetClipboardText(void*, char const*)>, ClipboardUserData = 0x55555572a240, ImeSetInputScreenPosFn = 0x7ffff7986e19
     <ImeSetInputScreenPosFn_DefaultImpl(int, int)>, ImeWindowHandle = 0x0, MousePos = {x = 1260, y = -83}, MouseDown = {false, false, false, false, false}, MouseWheel = 0, MouseWheelH = 0, KeyCtrl = false, 
[... cut ...]
          KeysDownDuration = {-1 <repeats 512 times>}, KeysDownDurationPrev = {-1 <repeats 512 times>}, NavInputsDownDuration = {-1 <repeats 21 times>}, NavInputsDownDurationPrev = {-1 <repeats 21 times>},
                    PenPressure = 0, InputQueueSurrogate = 0, InputQueueCharacters = {Size = 0, Capacity = 0, Data = 0x0}}
        _maximized = false
        fsWidth = 0
        fsHeight = 0
        fsPosX = 0
        fsPosY = 0
#17 0x0000555555654c23 in main (argc=1, argv=0x7fffffffdc98) at /root/SDRPlusPlus/src/main.cpp:5
No locals.

AlexandreRouma, 2021-09-24:
Ok thanks, I think that's all the info I'll need but the crash is very weird, seems to happen in the libc itself...

9a7aof, 2021-09-29:
Hi! 
I also run  gdb and sdrpp_ubuntu_focal_DEBUG_amd64.deb:

root@lenovo:/home/darko/src# dpkg -i sdrpp_ubuntu_focal_DEBUG_amd64.deb 
Odabir prije neodabranog paketa sdrpp.
(Čitanje baze podataka... 277700 datoteka ili direktorija su trenutno instalirana.)
Pripremanje za raspakiravanje sdrpp_ubuntu_focal_DEBUG_amd64.deb ...
Raspakiravam sdrpp (1.0.4-DEBUG) ...
Postavljam sdrpp (1.0.4-DEBUG) ...
Obrađujem signal za gnome-menus (3.36.0-1ubuntu1) ...
Obrađujem signal za desktop-file-utils (0.24-1ubuntu3) ...
Obrađujem signal za mime-support (3.64ubuntu1) ...
root@lenovo:/home/darko/src# exit
exit
darko@lenovo:~$ gdb /usr/bin/sdrpp 
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/sdrpp...
(gdb) bt full
No stack.
(gdb) run
Starting program: /usr/bin/sdrpp 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2021-09-29 08:06:04.162] [info] SDR++ v1.0.4
[2021-09-29 08:06:04.164] [info] Loading config
[New Thread 0x7fffed8f4700 (LWP 7046)]
[2021-09-29 08:06:04.166] [warning] ConfigManager locked, waiting...
[2021-09-29 08:06:04.469] [info] Using OpenGL 3.0

Thread 1 "sdrpp" received signal SIGSEGV, Segmentation fault.
__rawmemchr_sse2 () at ../sysdeps/x86_64/multiarch/../rawmemchr.S:37
37	../sysdeps/x86_64/multiarch/../rawmemchr.S: Nema takve datoteke ili direktorija.
(gdb) 
Thanks!

9a7aof, 2021-09-29:
Here is "my" build: 
darko@lenovo:~/src/SDRPlusPlus/build$ sudo su
root@lenovo:/home/darko/src/SDRPlusPlus/build# apt purge sdrpp

root@lenovo:/home/darko/src/SDRPlusPlus/build# sdrpp
Naredba 'sdrpp' nije pronađena, jeste li mislili:

root@lenovo:/home/darko/src/SDRPlusPlus/build# make install
[ 54%] Built target sdrpp_core
[ 56%] Built target sdrpp
[ 56%] Built target do_always
[ 58%] Built target airspy_source
[ 61%] Built target airspyhf_source
[ 63%] Built target file_source
[ 65%] Built target rtl_sdr_source
[ 67%] Built target rtl_tcp_source
[ 71%] Built target spyserver_source
[ 73%] Built target audio_sink
[ 75%] Built target network_sink
[ 77%] Built target new_portaudio_sink
[ 80%] Built target meteor_demodulator
[ 82%] Built target radio
[ 84%] Built target weather_sat_decoder
[ 91%] Built target discord-rpc
[ 93%] Built target discord_integration
[ 95%] Built target frequency_manager
[ 97%] Built target recorder
[100%] Built target rigctl_server
Install the project...
-- Install configuration: ""
-- Installing: /usr/bin/sdrpp
-- Set runtime path of "/usr/bin/sdrpp" to ""
-- Installing: /usr/share/sdrpp/bandplans
-- Installing: /usr/share/sdrpp/bandplans/france.json
-- Installing: /usr/share/sdrpp/bandplans/germany.json
-- Installing: /usr/share/sdrpp/bandplans/russia.json
-- Installing: /usr/share/sdrpp/bandplans/china.json
-- Installing: /usr/share/sdrpp/bandplans/general.json
-- Installing: /usr/share/sdrpp/bandplans/usa.json
-- Installing: /usr/share/sdrpp/bandplans/germany-mobile-lte-bands.json
-- Installing: /usr/share/sdrpp/bandplans/germany-mobile-networks.json
-- Installing: /usr/share/sdrpp/bandplans/canada.json
-- Installing: /usr/share/sdrpp/colormaps
-- Installing: /usr/share/sdrpp/colormaps/classic.json
-- Installing: /usr/share/sdrpp/colormaps/turbo.json
-- Installing: /usr/share/sdrpp/colormaps/gqrx.json
-- Installing: /usr/share/sdrpp/colormaps/inferno.json
-- Installing: /usr/share/sdrpp/colormaps/magma.json
-- Installing: /usr/share/sdrpp/colormaps/plasma.json
-- Installing: /usr/share/sdrpp/colormaps/classic_green.json
-- Installing: /usr/share/sdrpp/colormaps/websdr.json
-- Installing: /usr/share/sdrpp/colormaps/viridis.json
-- Installing: /usr/share/sdrpp/colormaps/electric.json
-- Installing: /usr/share/sdrpp/colormaps/greyscale.json
-- Installing: /usr/share/sdrpp/fonts
-- Installing: /usr/share/sdrpp/fonts/Roboto-Medium.ttf
-- Installing: /usr/share/sdrpp/icons
-- Installing: /usr/share/sdrpp/icons/stop.png
-- Installing: /usr/share/sdrpp/icons/sdrpp.ico
-- Installing: /usr/share/sdrpp/icons/menu.png
-- Installing: /usr/share/sdrpp/icons/center_tuning.png
-- Installing: /usr/share/sdrpp/icons/unmuted.png
-- Installing: /usr/share/sdrpp/icons/play.png
-- Installing: /usr/share/sdrpp/icons/sdrpp.png
-- Installing: /usr/share/sdrpp/icons/muted.png
-- Installing: /usr/share/sdrpp/icons/normal_tuning.png
-- Installing: /usr/share/sdrpp/themes
-- Installing: /usr/share/sdrpp/themes/grey.json
-- Installing: /usr/share/sdrpp/themes/light.json
-- Installing: /usr/share/sdrpp/themes/dark.json
[... 49 lines cut ...]

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/sdrpp...
(No debugging symbols found in /usr/bin/sdrpp)
(gdb) bt full
No stack.
(gdb) run
Starting program: /usr/bin/sdrpp 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2021-09-29 08:20:17.097] [info] SDR++ v1.0.4
[2021-09-29 08:20:17.098] [info] Loading config
[New Thread 0x7fffedd92700 (LWP 7744)]
[2021-09-29 08:20:17.589] [info] Using OpenGL 3.0

Thread 1 "sdrpp" received signal SIGSEGV, Segmentation fault.
__rawmemchr_sse2 () at ../sysdeps/x86_64/multiarch/../rawmemchr.S:37
37	../sysdeps/x86_64/multiarch/../rawmemchr.S: Nema takve datoteke ili direktorija.
(gdb) 

----------------------
The error is the same as with the original.

Thanks!

blacksmock, 2021-10-15:
I have experienced this fault on Lubuntu 20.04.3 LTS using a Hong-Kong manufactured Intel NUC clone and at present using Kernel 5.11.0.37-generic. SDR++ worked perfectly when I first tried it during August, but this was before Ubuntu  20.04 changed from using 5.8 series kernels with the latest point release. I tried re-installing the last 5.8 series kernel but this didn't change things . Some time ago I had installed Xubuntu 20.04 LTS onto a memory stick in order to see if I preferred the Xfce desktop. I installed SDR++ 1.0.3 on this installation,using the .deb package, which I booted from a USB 3 port on my machine. SDR++ worked perfectly with my Airspy HF+. This was using Kernel 5.4.0-26. Xubuntu then performed a large upgrade and installed Kernel 5.4.0-88 and SDR++ no longer worked after this upgrade, giving the segmentation fault when loading the Radio module. I rebooted Xubuntu with the old Kernel 5.4.0-26 from the Grub2 menu and SDR++ was again functioning correctly. There are a lot of kernels between 5.4.0-26 and 88, so tracking down what has changed might prove difficult. I could try some earlier versions of Kernel 5.8 which were current at the beginning of August, as SDR++ was definitely working on my hard disk when I first installed it. Hope this info helps and doesn't confuse the issue.

blacksmock, 2021-10-16:
After trying many different kernels on my Xubuntu 20.04.3 LTS memory stick installation, I reached the conclusion that SDR++ worked with kernels up to 5.4.0-86 and only failed with the current kernel 5.4.0-88-generic. There was no audio on these working instances, but SDR++ loaded correctly. The fault symptoms using kernel 5.4.0-88-generic had altered slightly. On loading the Radio module the program stopped then produced the error message "could not get server" then after nearly a minute failed with the segmentation fault as before. I noticed that once, when I had two instances of SDR++ open , the second one worked correctly after initially stalling at the loading Radio module stage. This was after I had opened the Synaptic package manager. As a last resort, I deleted the ~/ rtaudio directory which had been created when carrying out fix #121 and then re-compiled rtaudio again as per fix #121. This restored SDR++ to full working order and my audio sink destinations were now visible. I also did this on my Lubuntu 20.04.3 hard disk installation, using 5.11 series kernel, which also restored SDR++ to normal operation.

closed 2022-09-29 by AlexandreRouma