AlexandreRouma/SDRPlusPlus #322: Select Active Radio

rs1729, 2021-08-28:
Multiple radios works fine.
Suppose there are two radios (named Radio/VCO1 in the picture) at the same frequency. I can deselect and select each such that it becomes active in terms of frequency change (and audio mute) at the top. I can mute each audio also on the left site, and I can select the radio by clicking into the filter window. However if e.g. the RAW filter window covers the FM filter at the same frequency, the only way I found is deselecting the FM radio and selecting it again, so I can change the frequency of  the FM radio ("VCO1" in the picture). Or I could move the RAW radio frequency such that I can reach the smaller window behind it.
Is there a way to choose the radio for which I want to change the frequency? And maybe it could be shown somewhere which radio is "active", i.e. to which radio the frequency display on top belongs?
[image]

Don't know if it is possible to have an optional IF spectrum or the IQ spectrum of the selected filter below the waterfall. Perhaps there is not enough space for multiple IF windows for each radio, but maybe the "active" radio IF spectrum could be displayed. (helps fine tuning without zooming in)

Btw. I believe, selecting/deselecting a radio with Source:File makes the program freeze/wait.

AlexandreRouma, 2021-08-28:
Hello,

> Is there a way to choose the radio for which I want to change the frequency?
You use the Page Up and Page Down keys to switch through VFOs

> And maybe it could be shown somewhere which radio is "active"
The currently selected radio has the tuning line colored red.

rs1729, 2021-08-28:
Thanks. This should work for me.

Off topic, about the "IF spectrum" feature, is this something that was requested before, is this something on your list?

closed 2021-08-28 by rs1729

AlexandreRouma, 2021-08-28:
Not really.
To be honest I don't really see the point, the IF spectrum will be exactly the same as what you see when zooming in on the VFO...

Now, audio spectrum, maybe, but I sitll have to figure out a clean way to show it. I don't like how GQRX shows it super small in a side menu, and I don't like how SDR# shows it at the bottom.

rs1729, 2021-08-28:
Often it is helpful (for signals that I'm interested in) to see the entire spectrum and also the spectrum of the filter width around a signal, so I can see the signal drifting.
I'm not interested in the (demodulated?) audio spectrum, but I cannot speak for others.
When zoomed in, I don't see if there are other signals popping up, and also the FFT resolution would have to be increased. Or do you resample when zooming in? The IF spectrum is what I like about SDR#, SDRConsole is ok, but different concept I guess, and both for Windows.
Anyway, sdr++ looks very promising.