AlexandreRouma/SDRPlusPlus #326: filename of recording

rs1729, 2021-08-29:
Is it possible to change date/time format of the filename of a recording to YYYY-MM-DD_hh-mm-ss? Would be more appropriate for sorting and more in line with ISO.
And for "Audio" recordings, maybe the chosen modulation (e.g. NFM, RAW/IQ, AM, etc). could be added to the filename?
Or is there a way to configure this in the config files?

AlexandreRouma, 2021-08-29:
For sorting by date, use your filemanager's built-in "Sort by date" function.
The YYYY-MM-DD is vert annoying to read for most people around the world.

As for adding the mode in the filename, I guess it could be done but I'm not sure how useful that would be as information.

There is currently no way to change the format without editing the source code, but if enough people care about it I might consider adding an option.

rs1729, 2021-08-30:
> For sorting by date, use your file manager's built-in "Sort by date" function.

Sorry, but no, if I have a list of filenames, I cannot use file metadata.

> The YYYY-MM-DD format is very annoying to read for most people around the world.

Didn't hear that before, but the problem of dates like 01-09-2021 is that it is ambiguous around the world.
(Although you could apply your use-metadata argument to both formats I guess.)
I believe one of the ideas behind ISO 8601 (and ISO 2014) is to have an unambiguous date format.
Ordering by most-to-least-significant should be easy to understand in the sdr world.

closed 2022-08-17 by AlexandreRouma

beta-tester, 2022-08-17:
is ISO format implemented now?

AlexandreRouma, 2022-08-17:
Implementation is not planned

beta-tester, 2022-08-18:
> Implementation is not planned

that's very disappointing

1. that not following international standards
2. that this issue was closed without telling the reason why