AlexandreRouma/SDRPlusPlus #333: [Feature] Avoid aliased regions when free-tuning

shuyuan-liu, 2021-08-31:
For my RTL-SDR and Airspy Mini, roughly the lowest and highest 10% of the spectrum are aliased. Tuning to a station that happens to be in these regions in free-tuning mode means reception would not be optimal. The following screenshots show the same station when placed in the high end of the spectrum and when placed in the middle; the SNR is about 20dB lower in the former case.

[image]

[image]

My idea is that aliased regions should be automatically avoided in free-tuning mode. For example, when the user scrolls the VFO up to the 90% point of the spectrum, a retune happens, shifting the spectrum so that the VFO is placed at the 10% point, staying within the non-aliased region.

AlexandreRouma, 2021-08-31:
This is already partially the case. When tuning out of range the VFO is put to 10% above or below the edge to avoid ending up in these regions.
Unfortunately, this limits depends a lot of the SDR or specific baseband recording so not allowing tuning through scroll would not be acceptable.

The way SDR# deals with this is to straight up cut the bandwidth for **all** SDRs and not even allow seeing the aliased areas. This is flawed as quite a few SDRs have a full alias free bandwidth.

In my opinion it's better to let the user tune anywhere they want and consider the entire spectrum as valid. If a user wants to receive those areas, they can simply drag the frequency scale a bit to bring it into the valid receiving part of the spectrum.

shuyuan-liu, 2021-08-31:
Thanks for the explanation @AlexandreRouma, I see your point. I'll close this issue for now and reopen if I've got anything else to add.

closed 2021-08-31 by shuyuan-liu