AlexandreRouma/SDRPlusPlus #334: Mac build missing SDR Play

ve3ds, 2021-09-01:
Likely this is being fixed, but the MacOS version is missing the SDRPlay  drivers ... will they be there in 1.04? Great program and REALLY needed on MacOS ... really really needed. Thx DS

AlexandreRouma, 2021-09-01:
Unless SDRplay adds a way to install the library unattended, I can't include it in the prebuilt mac version.
Anyone who wishes to use SDR++ with an SDRplay SDR on mac currently has to build it themselves following the linux instructions.

ve3ds, 2021-09-01:
Alex Ok, I guess I’ll give it a try!! 
I’ve loaded all the dependencies to get the pre build Mac to run.  What do i need for SDRPlay? Soapy, or the proprietary drivers?
Thanks
73 DS

AlexandreRouma, 2021-09-01:
You need to install the API/drivers from SDRplay's website, then you should be able to use the following commands to build and install SDR++.

First, install the dependencies;
```sh
brew update
brew install fftw glew glfw volk airspy airspyhf portaudio hackrf rtl-sdr libbladerf
```

Then download the SDR++ source code:
```sh
git clone https://github.com/AlexandreRouma/SDRPlusPlus
```

Prepare the build:
```sh
cd SDRPlusPlus
mkdir build
cd build
cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON
```

Build and install:
```sh
make
sudo make install
```

ve3ds, 2021-09-01:
Ok… looks like I missed a few dependencies after all !

Thanks I’ll give it a go.

Thanks 
Dana VE3DS
in Toronto.

ve3ds, 2021-09-01:
I keep getting an error …

cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON    

-- Checking for module 'glew'
--   No package 'glew' found
CMake Error at /opt/local/share/cmake-3.17/Modules/FindPkgConfig.cmake:497 (message):
  A required package was not found

==> Pouring glew--2.2.0_1.big_sur.bottle.tar.gz
🍺  /usr/local/Cellar/glew/2.2.0_1: 38 files, 3.4MB

Looks like its not seeing the path or version of cmake? 
Any suggestions?

Dana

AlexandreRouma, 2021-09-01:
I'm not sure.
I don't have access to a mac so I can't try anything but it seems to happen often with people on Big Sur.
For some reason homebrew installs it somewhere and pkgconfig can't find it...

ve3ds, 2021-09-01:
Ok …. Im trying to find the correct path… 
I’ll track it down. Saw your comment about not having access to a mac.
I’ll circle back if i figure it out.

DS

orryverducci, 2021-09-02:
> Unless SDRplay adds a way to install the library unattended, I can't include it in the prebuilt mac version.
> Anyone who wishes to use SDR++ with an SDRplay SDR on mac currently has to build it themselves following the linux instructions.

The installer provided by SDRplay is a standard package installer. As such you should be able to install it unattended with logging using the following command: `installer -allowUntrusted -verbose -pkg "/path/to/pkg/SDRplay_RSP_API-MacOSX-3.07.3.pkg"`

loewal, 2021-09-02:
> Ya can’t get it to make… keeps throwing not being able to find volk now… looks like the code isnt pointing properly..
> Same issue with older macOS as well.
> Will keep working on it…
> Anyone have success with Big Sur ?
>
> Dana

Working fine here.

Grtz.

loewal, 2021-09-02:
> Grtz.
> can you share the procedure to build that you used?
>
> Dana

I’m sorry, but that wat some months ago.
As always is use Homebrew (brew) to install missing dependencies.
Also had to comment some entries, like Perseus etc.

Succes!

Loe

syepes, 2021-09-03:
Just tried it with 11.5.2 (20G95) and it did not build.

ve3ds, 2021-09-03:
Yes that’s my issue with 11.15.2 
Can’t find some of the casks… not sure if its a path issue or what.
GLFW3 or GLFW?
If I run BREW LIST its shown, but the build says it can’t find GLFW3
So is it syntax or path or just broken?

Dana

ve3ds, 2021-09-03:
from what I see GLFW3 is now just GLFW

so why is cmake looking for GLFW3?

Dana

AlexandreRouma, 2021-09-03:
the lib is called GLFW, not GLFW3
It has the same name on all operating systems.

What I suspect is that in newer versions brew doesn't install the .pc file, correctly and pkg-config can't find the library

ericjjjohnson, 2021-09-05:
trying to build on an m1 silicon mac.   hombrew for m1 installed get:
🍺  /opt/homebrew/Cellar/glfw/3.3.4: 14 files, 540.8KB
Error: volk: no bottle available!
You can try to install from source with:
  brew install --build-from-source volk
Please note building from source is unsupported. You will encounter build
failures with some formulae. If you experience any issues please create pull
requests instead of asking for help on Homebrew's GitHub, Twitter or any other
official channels.

This results in ==> Installing volk
==> python3 -m venv --system-site-packages /opt/homebrew/Cellar/volk/2.5.0/libex
==> /opt/homebrew/Cellar/volk/2.5.0/libexec/venv/bin/pip install -v --no-deps --
==> cmake . -DPYTHON_EXECUTABLE=/opt/homebrew/Cellar/volk/2.5.0/libexec/venv/bin
==> make install
Last 15 lines from /Users/ejj3/Library/Logs/Homebrew/volk/04.make:
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib -I/tmp/volk-20210905-16784-b6f54/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_malloc.c.o -MF CMakeFiles/volk_obj.dir/volk_malloc.c.o.d -o CMakeFiles/volk_obj.dir/volk_malloc.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk_malloc.c
[ 82%] Building C object lib/CMakeFiles/volk_obj.dir/volk.c.o
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DLV_MACHINE_GENERIC_ORC -DLV_MACHINE_NEONV8 -DLV_MACHINE_NEON_ORC -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib -I/tmp/volk-20210905-16784-b6f54/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk.c.o -MF CMakeFiles/volk_obj.dir/volk.c.o.d -o CMakeFiles/volk_obj.dir/volk.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk.c
[ 83%] Building C object lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DVOLK_CPU_FEATURES=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib -I/tmp/volk-20210905-16784-b6f54/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o -MF CMakeFiles/volk_obj.dir/volk_cpu.c.o.d -o CMakeFiles/volk_obj.dir/volk_cpu.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk_cpu.c
[ 84%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machines.c.o
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DLV_MACHINE_GENERIC_ORC -DLV_MACHINE_NEONV8 -DLV_MACHINE_NEON_ORC -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib -I/tmp/volk-20210905-16784-b6f54/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_machines.c.o -MF CMakeFiles/volk_obj.dir/volk_machines.c.o.d -o CMakeFiles/volk_obj.dir/volk_machines.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk_machines.c
[ 85%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_generic_orc.c.o
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib -I/tmp/volk-20210905-16784-b6f54/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_machine_generic_orc.c.o -MF CMakeFiles/volk_obj.dir/volk_machine_generic_orc.c.o.d -o CMakeFiles/volk_obj.dir/volk_machine_generic_orc.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk_machine_generic_orc.c
[ 86%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o
cd /tmp/volk-20210905-16784-b6f54/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20210905-16784-b6f54/include -I/tmp/volk-20210905-16784-b6f54/kernels -I/tmp/volk-20210905-16784-b6f54/lib - [... cut ...] ch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk -fPIC -std=gnu11 -funsafe-math-optimizations -MD -MT lib/CMakeFiles/volk_obj.dir/volk_machine_neon_orc.c.o -MF CMakeFiles/volk_obj.dir/volk_machine_neon_orc.c.o.d -o CMakeFiles/volk_obj.dir/volk_machine_neon_orc.c.o -c /tmp/volk-20210905-16784-b6f54/lib/volk_machine_neon_orc.c
[ 87%] Built target volk_obj
make: *** [all] Error 2

It would be loverly to have a complete release available. sorry you don't have access to a mac.

ve3ds, 2021-09-05:
I used ibrew to solve the missing volk … but yes frustrating with the Mac…
Should we try building it on the version prior to Big Sur??
Or can someone make a build for non M1 and M1 with the SDRPlay incorporated ?
GLFW3 … what to do… 

Anyway looks really good and the HackRF version runs … too bad you cant just point to the file to run
SDRPlay… humbug on proprietary stuff eh?

Cheers 
Dana VE3DS

KahikoHonu, 2021-09-06:
Just to let you know, I was able to successfully build, and run, SDR++ on macOS Monterey (beta), and use my RSPdx from SDRPlay, using only a few minor changes from the instructions here.

My first change, after making and cd'ing into the build directory was to use following for the cmake command:

`cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_AIRSPY_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON`

I know that SDRPlay should be a default enabled source, but, I wanted to make sure it was enabled, and, I had to explicitly disable Airspy as a source.  There were complaints during compilation regarding building the airspy module.  I don't have one, so, I didn't bother trying to fix.

The aforementioned "brew" commands took care of installing everything else.

From there, it was as easy as compiling it, making sure that the airspy modules were disabled in the JSON config file (which I did earlier when trying to get this to work with the macOS PKG version of SDR++ (did not work with SDRPlay), and running it.

Selected my SDRPlay source, and immediately had audio working fine.

AlexandreRouma, 2021-09-06:
> I know that SDRPlay should be a default enabled source

As noted in the big module list in the readme, the SDRplay source module is **NOT** enabled by default. This is because it requires a proprietary library that cannot be installed easily through a package manager.

> making sure that the airspy modules were disabled in the JSON config file 

This should not be needed, SDR++ will just display an error saying it can't load it and move on

KahikoHonu, 2021-09-06:
Sorry - should have been more clear - when I first tried to run sdrpp, with airspy enabled, and the macOS package, it immediately crashed.  I saw the discussion on this Issue here, so, decide to build from source.

Trying to do that failed as well, if I allowed OPT_BUILD_AIRSPY_SOURCE to be ON.

When I examined the CmakeCache, I saw that OPT_BUILD_SDRPLAY_SOURCE was set to ON, as was OPT_BUILD_AIRSPY_SOURCE.  

I then downloaded the SDRPlay API for macOS package, modified the cmake options, and it built.  

Could well be that the cmake command line args did not need my change, and having the SDRPlay API installed would have been picked up automatically - I was not expecting the cmake/installation of sdrpp to install the SDRPlay API.

The JSON tweaks were from my initial attempt to get sdrpp to run at all from the macOS package; as it was crashing as it was trying to initialise Airspy.  I had hoped simply disabling it in the JSON config would have allowed it to start and to work with my hackrf, it did not.

Anyway, just sharing what worked for me, looks great, having fun!  Thanks!

pmaine, 2021-09-06:
I have a mini Mac circa 2012 running Catalina.
Can you help me with all of the brew commands from start to finish to build on my Mac? 

I have successfully built on Ubuntu and Raspian.

KahikoHonu, 2021-09-06:
From my history file:

`
brew install fftw glew glfw volk airspy airspyhf portaudio hackrf rtl-sdr libbladerf
git clone https://github.com/AlexandreRouma/SDRPlusPlus
cd SDRPlusPlus
mkdir build
cd build
cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_AIRSPY_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON
make
sudo make install
`

And, as I said, I also had to install the proprietary SDRPlay API -- and I've used homebrew for quite some time, so, I honestly can't say if my success was in any way related to other packages I've coincidentally already installed.  Oh, and as I said, I'm on the macOS Beta, so, that may also be an issue.

KahikoHonu, 2021-09-06:
From my history file:

`
brew install fftw glew glfw volk airspy airspyhf portaudio hackrf rtl-sdr libbladerf

git clone https://github.com/AlexandreRouma/SDRPlusPlus

cd SDRPlusPlus

mkdir build

cd build

cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_AIRSPY_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON

make

sudo make install
`

And, as I said, I also had to install the proprietary SDRPlay API -- and I've used homebrew for quite some time, so, I honestly can't say if my success was in any way related to other packages I've coincidentally already installed.  Oh, and as I said, I'm on the macOS Beta, so, that may also be an issue.

(Sorry for the horrid formatting of the above)

AlexandreRouma, 2022-09-13:
was added a while back

closed 2022-09-13 by AlexandreRouma