AlexandreRouma/SDRPlusPlus #337: HackRF Crash

JosephTheTank13, 2021-09-04:
When I use the HackRF source, the program crashes with a segmentation fault when I hit refresh to search for devices. On restart of SDRPP, it shows the device string like it found the device, but if you press play the program crashes immediately. Using the SoapySDR source and choosing the HackRF kinda words, but it's really slow and choppy for some reason. Here's the attached terminal output.
`<pre><font color="#00CD00"><b>joseph@josephbeastpc</b></font>:<font color="#0000EE"><b>~</b></font>$ sdrpp`
`[2021-09-04 11:52:42.736] [<font color="#00CD00">info</font>] SDR++ v1.0.4`
`[2021-09-04 11:52:42.737] [<font color="#00CD00">info</font>] Loading config`
`[2021-09-04 11:52:42.737] [<font color="#CDCD00"><b>warning</b></font>] ConfigManager locked, waiting...`
`[2021-09-04 11:52:42.884] [<font color="#00CD00">info</font>] Using OpenGL 3.0`
`[2021-09-04 11:52:43.006] [<font color="#00CD00">info</font>] Loading icons`
`[2021-09-04 11:52:43.009] [<font color="#00CD00">info</font>] Loading band plans`
`[2021-09-04 11:52:43.016] [<font color="#00CD00">info</font>] Loading band plans color table`
`[2021-09-04 11:52:43.026] [<font color="#00CD00">info</font>] Loading modules`
`[2021-09-04 11:52:43.026] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so`
`[2021-09-04 11:52:43.032] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/file_source.so`
`[2021-09-04 11:52:43.038] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/audio_sink.so`
`[2021-09-04 11:52:43.046] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/spyserver_source.so`
`[2021-09-04 11:52:43.051] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/radio.so`
`[2021-09-04 11:52:43.058] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
`[2021-09-04 11:52:43.065] [<font color="#CDCD00"><b>warning</b></font>] ConfigManager locked, waiting...`
`[2021-09-04 11:52:43.065] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/rigctl_server.so`
`[2021-09-04 11:52:43.073] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so`
`[2021-09-04 11:52:43.079] [<font color="#CD0000"><b>error</b></font>] Couldn&apos;t load /usr/lib/sdrpp/plugins/sdrplay_source.so.`
`[2021-09-04 11:52:43.079] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/bladerf_source.so`
`[2021-09-04 11:52:43.086] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/network_sink.so`
`[2021-09-04 11:52:43.093] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so`
`[2021-09-04 11:52:43.100] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so`
`[2021-09-04 11:52:43.107] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so`
`[2021-09-04 11:52:43.114] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so`
`[2021-09-04 11:52:43.121] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so`
`[2021-09-04 11:52:43.128] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/recorder.so`
`[2021-09-04 11:52:43.135] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/airspy_source.so`
`[2021-09-04 11:52:43.142] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/soapy_source.so`
`[2021-09-04 11:52:43.149] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/discord_integration.so`
`[2021-09-04 11:52:43.157] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/hackrf_source.so`
`[2021-09-04 11:52:43.163] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so`
`[2021-09-04 11:52:43.169] [<font color="#00CD00">info</font>] Loading /usr/lib/sdrpp/plugins/limesdr_source.so`
`[2021-09-04 11:52:43.176] [<font color="#CD0000"><b>error</b></font>] Couldn&apos;t load /usr/lib/sdrpp/plugins/limesdr_source.so.`
`[2021-09-04 11:52:43.176] [<font color="#00CD00">info</font>] Initializing Airspy Source (airspy_source)`
`[2021-09-04 11:52:43.187] [<font color="#00CD00">info</font>] Initializing AirspyHF+ Source (airspyhf_source)`
`[2021-09-04 11:52:43.194] [<font color="#00CD00">info</font>] Initializing Audio Sink (audio_sink)`
`[2021-09-04 11:52:43.197] [<font color="#00CD00">info</font>] Initializing BladeRF Source (bladerf_source)`
`[2021-09-04 11:52:43.208] [<font color="#CD0000"><b>error</b></font>] Could not list devices`
`[2021-09-04 11:52:43.208] [<font color="#00CD00">info</font>] Initializing File Source (file_source)`
`[2021-09-04 11:52:43.211] [<font color="#00CD00">info</font>] Initializing Frequency Manager (frequency_manager)`
`[2021-09-04 11:52:43.218] [<font color="#00CD00">info</font>] Initializing HackRF Source (hackrf_source)`
`[2021-09-04 11:52:43.235] [<font color="#00CD00">info</font>] Initializing LimeSDR Source (limesdr_source)`
`[2021-09-04 11:52:43.236] [<font color="#CD0000"><b>error</b></font>] Module &apos;limesdr_source&apos; doesn&apos;t exist`
`[2021-09-04 11:52:43.236] [<font color="#00CD00">info</font>] Initializing PlutoSDR Source (plutosdr_source)`
`[2021-09-04 11:52:43.239] [<font color="#00CD00">info</font>] Initializing RTL-SDR Source (rtl_sdr_source)`
`[2021-09-04 11:52:43.249] [<font color="#00CD00">info</font>] Initializing RTL-TCP Source (rtl_tcp_source)`
`[2021-09-04 11:52:43.253] [<font color="#00CD00">info</font>] Initializing Radio (radio)`
`[2021-09-04 11:52:43.572] [<font color="#00CD00">info</font>] RtAudio stream open`
`[2021-09-04 11:52:43.572] [<font color="#00CD00">info</font>] Initializing Recorder (recorder)`
`[2021-09-04 11:52:43.573] [<font color="#00CD00">info</font>] Initializing Rigctl Server (rigctl_server)`
`[2021-09-04 11:52:43.579] [<font color="#00CD00">info</font>] Initializing SDRplay Source (sdrplay_source)`
`[2021-09-04 11:52:43.586] [<font color="#CD0000"><b>error</b></font>] Module &apos;sdrplay_source&apos; doesn&apos;t exist`
`[2021-09-04 11:52:43.586] [<font color="#00CD00">info</font>] Initializing SoapySDR Source (soapy_source)`
`[INFO] Opening HackRF One #0 75b068dc313b6807...`
`[2021-09-04 11:52:43.664] [<font color="#00CD00">info</font>] Setting sample rate to 20000000.0`
`[2021-09-04 11:52:43.664] [<font color="#00CD00">info</font>] New DSP samplerate: 20000000.0 (source samplerate is 20000000.0)`
`[2021-09-04 11:52:43.664] [<font color="#00CD00">info</font>] Updating FFT DSP settings: Keep: 65536, Skip: 934464`
`[2021-09-04 11:52:43.683] [<font color="#00CD00">info</font>] Initializing SpyServer Source (spyserver_source)`
`[2021-09-04 11:52:43.690] [<font color="#00CD00">info</font>] Loading color maps`
`[2021-09-04 11:52:43.697] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/viridis.json`
`[2021-09-04 11:52:43.704] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/turbo.json`
`[2021-09-04 11:52:43.711] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/websdr.json`
`[2021-09-04 11:52:43.718] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/gqrx.json`
`[2021-09-04 11:52:43.725] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/classic_green.json`
`[2021-09-04 11:52:43.732] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/classic.json`
`[2021-09-04 11:52:43.739] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/magma.json`
`[2021-09-04 11:52:43.746] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/plasma.json`
`[2021-09-04 11:52:43.753] [<font color="#00CD00">info</font>] Loading /usr/share/sdrpp/colormaps/greyscale.json`
[... 4 lines cut ...]
`[2021-09-04 11:52:43.803] [<font color="#00CD00">info</font>] HackRFSourceModule &apos;HackRF Source&apos;: Menu Select!`
`[2021-09-04 11:52:43.803] [<font color="#00CD00">info</font>] New DSP samplerate: 2000000.0 (source samplerate is 2000000.0)`
`[2021-09-04 11:52:43.803] [<font color="#00CD00">info</font>] Updating FFT DSP settings: Keep: 65536, Skip: 34464`
`[2021-09-04 11:52:43.811] [<font color="#00CD00">info</font>] RtAudio stream open`
`[2021-09-04 11:52:43.848] [<font color="#00CD00">info</font>] Updating FFT DSP settings: Keep: 65536, Skip: 34464`
`[2021-09-04 11:52:43.848] [<font color="#00CD00">info</font>] Updating FFT DSP settings: Keep: 65536, Skip: 34464`
[... cut ...]
`[2021-09-04 11:52:43.852] [<font color="#00CD00">info</font>] Running post-init for Recorder`
`[2021-09-04 11:52:43.852] [<font color="#00CD00">info</font>] Running post-init for Rigctl Server`
`[2021-09-04 11:52:43.852] [<font color="#00CD00">info</font>] Running post-init for SoapySDR Source`
`[2021-09-04 11:52:43.852] [<font color="#00CD00">info</font>] Running post-init for SpyServer Source`
`[2021-09-04 11:52:43.852] [<font color="#00CD00">info</font>] Ready.`
`Segmentation fault (core dumped)`

JosephTheTank13, 2021-09-04:
oh gosh the terminal output looks like trash

AlexandreRouma, 2021-09-04:
See the [troubleshooting section](https://github.com/AlexandreRouma/SDRPlusPlus#sdr-crashes-when-starting-a-hackrf) of the readme:
[image]

JosephTheTank13, 2021-09-04:
what do I do if there is no /plugins/ directory?

AlexandreRouma, 2021-09-05:
That's not possible, the logs literally say they're loading modules from there...

JosephTheTank13, 2021-09-05:
`/usr/share/sdrpp$ ls`
`bandplans  colormaps  fonts  icons  themes`

JosephTheTank13, 2021-09-05:
Oh it's loading from /usr/lib/sdrpp/plugins. Removing the "Soapy Source" option inside sdrpp and restarting fixed it for me. It doesn't crash now, but it barely works. The waterfall is slow and choppy, and changing the bandwidth makes it completely stop working until I restart SDR++.

BloodBlight, 2021-09-27:
When I did this, it just re-created the file.  Interesting though, it's a different size:
```
-rw-r--r-- 1 root root  555904 Aug  9 16:01 soapy_source.so
-rw-r--r-- 1 root root  533552 Aug  9 16:01 soapy_source.so_back
...
md5sum soapy_source.*
4a8e17f13cf301f2d30212d5fc78e3ef  soapy_source.so
7537d3656b82d35a61c21243f0d0ebeb  soapy_source.so_back
```

BloodBlight, 2021-09-27:
Moved both files out of the folder and it worked....  Weird...

editorialz, 2023-01-07:
Do a search in root for "soapy_source.so" in Linux if there's no plugin folder. I found it in libs, deleted it and that fixed the problem.

CTI-Tim, 2023-06-05:
This bug is still happening 2 years later.   anyone know when the lib for hackRF will be fixed or is even on the roadmap to get fixed?