AlexandreRouma/SDRPlusPlus #338: Crashing/Closing early in startup process (on Initializing Recorder?)

grbman, 2021-09-06:
Hi,
I'm attempting to run SDR++ 1.0.3 on Windows 10 Version 10.0.19042 Build 19042 and I'm not able to get through the startup process. Running sdrpp.exe -s gives me: 
```sdrpp.exe -s
[2021-09-05 17:30:37.202] [info] SDR++ v1.0.3
[2021-09-05 17:30:37.204] [info] Loading config
[2021-09-05 17:30:37.908] [info] Loading icons
[2021-09-05 17:30:37.927] [info] Loading band plans
[2021-09-05 17:30:37.933] [info] Loading band plans color table
[2021-09-05 17:30:37.939] [info] Loading modules
[2021-09-05 17:30:37.940] [info] Loading ./modules/airspyhf_source.dll
[2021-09-05 17:30:37.947] [info] Loading ./modules/airspy_source.dll
[2021-09-05 17:30:37.951] [info] Loading ./modules/audio_sink.dll
[2021-09-05 17:30:37.956] [info] Loading ./modules/bladerf_source.dll
[2021-09-05 17:30:37.960] [info] Loading ./modules/discord_integration.dll
[2021-09-05 17:30:37.963] [info] Loading ./modules/file_source.dll
[2021-09-05 17:30:37.968] [info] Loading ./modules/frequency_manager.dll
[2021-09-05 17:30:37.970] [info] Loading ./modules/hackrf_source.dll
[2021-09-05 17:30:37.978] [info] Loading ./modules/limesdr_source.dll
[2021-09-05 17:30:37.983] [info] Loading ./modules/meteor_demodulator.dll
[2021-09-05 17:30:37.987] [info] Loading ./modules/network_sink.dll
[2021-09-05 17:30:37.990] [info] Loading ./modules/plutosdr_source.dll
[2021-09-05 17:30:37.995] [info] Loading ./modules/radio.dll
[2021-09-05 17:30:37.998] [info] Loading ./modules/recorder.dll
[2021-09-05 17:30:38.001] [info] Loading ./modules/rigctl_server.dll
[2021-09-05 17:30:38.004] [info] Loading ./modules/rtl_sdr_source.dll
[2021-09-05 17:30:38.007] [info] Loading ./modules/rtl_tcp_source.dll
[2021-09-05 17:30:38.013] [info] Loading ./modules/sdrplay_source.dll
[2021-09-05 17:30:38.020] [info] Loading ./modules/soapy_source.dll
[2021-09-05 17:30:38.024] [info] Loading ./modules/spyserver_source.dll
[2021-09-05 17:30:38.028] [info] Initializing Airspy Source (airspy_source)
[2021-09-05 17:30:38.052] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-09-05 17:30:38.075] [info] Initializing Audio Sink (audio_sink)
[2021-09-05 17:30:38.078] [info] Initializing HackRF Source (hackrf_source)
[2021-09-05 17:30:38.103] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-09-05 17:30:38.105] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-09-05 17:30:38.106] [info] Initializing Radio (radio)
[2021-09-05 17:30:38.205] [info] RtAudio stream open
[2021-09-05 17:30:38.206] [info] Initializing Recorder (recorder)
```
This is my recorder_config.json:
```
{
    "Recorder": {
        "directory": "%ROOT%/recordings",
        "recMode": 1
    }
}
```
I do have a recordings folder but it's empty.

Any recommendations to get me running?

AlexandreRouma, 2021-09-06:
That config file looks like it's from a really old version, did you update from one of the beta versions?
If yes, install SDR++ in a clean directory.

grbman, 2021-09-06:
Hi @AlexandreRouma ,
Yes I did have a beta version installed previously. I deleted the old instance and installed the latest version, and this solved my issue.

Thanks for the quick response!

closed 2021-09-06 by grbman