AlexandreRouma/SDRPlusPlus #340: Package for ARM64?

JohnDusek3, 2021-09-07:
Wrong place to enter this request, but I can't find a better one.

Is it possible to get this package to run on ARM64?

I am using Ubuntu on Raspberry Pi 4 (8GB).

AlexandreRouma, 2021-09-07:
yes, just compile and install using the linux instructions.

pmaine, 2021-09-07:
Yes, I have built and am running on my Pi4 8GB.

What SDRs do you plan to use?

JohnDusek3, 2021-09-07:
Airspy HF+ Discovery, and RTLSDR

Do you have a compile already completed that I could use?

pmaine, 2021-09-07:
I don't have an easy way to get it to you. I will pull my notes together and add a comment in a subsequent post that lists all of the steps to build for a Pi.

JohnDusek3, 2021-09-07:
That would be GREAT.  

I am a relatively new Linux user and can follow steps at the command line, but what dependencies to check for, how to check for them, and where to put dependencies I need explicit direction on.  

From the notes it looks like Airspy and RTL-SDR dependencies are included with the base, so that's positive.

pmaine, 2021-09-07:
What is the exact version of Ubuntu you are using on your Pi?

I am running Raspian OS but I’ll install the same image as you to give you all of the commands.

JohnDusek3, 2021-09-07:
Hi -

I am using Ubuntu 21.04 64-bit (which includes Gnome 3.38.5), and standard config defaults.

John

AlexandreRouma, 2021-09-07:
That won't work, you can't copy binaries from armbian to ubuntu 21.04. And you sitll have to install the dependencies anyway.

Here are the commands to build and install:

```sh
sudo apt install build-essential cmake git libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libairspyhf-dev libairspy-dev librtaudio-dev librtlsdr-dev
git clone https://github.com/AlexandreRouma/SDRPlusPlus
cd SDRPlusPlus
mkdir build
cd build
cmake .. -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=OFF -DOPT_BUILD_HACKRF_SOURCE=OFF
make
sudo make install
```

JohnDusek3, 2021-09-07:
Thank you – I will try these steps.

alphafox02, 2021-09-07:
Here’s an aarch64 Pi image with everything Pre installed. Not persuading anyone to use it vs building yourself, only sharing as a suggestion. 

https://sourceforge.net/projects/dragonos-pi64/files/

JohnDusek3, 2021-09-08:
Hi -  thank you.

What distribution is DragonOS based on, Ubuntu?

https://sourceforge.net/projects/dragonos-pi64/files/

alphafox02, 2021-09-08:
20.04 Ubuntu with some SDR++ prerequisites being installed from source.

ericek111, 2021-09-09:
Works great on my Samsung S10e in Proot in Termux.

JohnDusek3, 2021-09-09:
Works great on my RPi4 8GB also!

SDR++ started, recognized my Airspy HF+ Discovery (after the appropriate setting) and worked great.

Thank you, Alphafox02 for the suggestion…

Only issue I have is Lubuntu/DragonOS doesn’t have a good software package manager in the GUI like Ubuntu, that seems to be by design.

yosimel, 2021-09-17:
> 20.04 Ubuntu with some SDR++ prerequisites being installed from source.

Hi
so if i understand right... the distro you make is including Sdrtrunk?
is it working? including the java package and all extra to support it?
i plan on using your distro on my new Raspberry pi4 8G ram
will it work?

alphafox02, 2021-09-17:
That is correct. The distro for the pi includes sdr++. Just not I build it and install it from source with it accounting for sdrplay, bladerf, LimeSDR. So if you want to upgrade. I suggest doing it from source like I do. 

Sdrtrunk with the proper Java is included as well. You’ll need at least an internet connection to go in and pull down I think the decoder for p25 etc which it does it all for you. Sorry, don’t want to get too far off topic. Pi build w/ sdr++ etc is here and please refer to readme on the desktop for post install tips 
https://sourceforge.net/projects/dragonos-pi64/files/

yosimel, 2021-09-18:
> That is correct. The distro for the pi includes sdr++. Just note I build it and install it from source with it accounting for sdrplay, bladerf, LimeSDR. So if you want to upgrade. I suggest doing it from source like I do.
> 
> Sdrtrunk with the proper Java is included as well and should be fine with your pi. You’ll need at least an internet connection to go in and pull down I think the decoder for p25 etc which it does it all for you. Sorry, don’t want to get too far off topic. Pi build w/ sdr++ etc is here and please refer to readme on the desktop for post install tips
> https://sourceforge.net/projects/dragonos-pi64/files/

Thank you for this..
I try this and it works..
I just dont know how to get Admin privliges...can not save changes of files inside directories...
All this apps you included in this dragonOS, could build on resbian OS or olny suite with ubuntu platform for pi?

AlexandreRouma, 2021-09-18:
Please move this conversation to the proper location for dragonos issue rather than this repository

yosimel, 2021-09-18:
You are right..I am sorry for this
-------- הודעה מקורית --------מאת: AlexandreRouma ***@***.***> תאריך: 18.9.2021  17:13  (GMT+02:00) אל: AlexandreRouma/SDRPlusPlus ***@***.***> עותק: yosimel ***@***.***>, Comment ***@***.***> נושא: Re: [AlexandreRouma/SDRPlusPlus] Package for ARM64? (#340) 
Please move this conversation to the proper location for dragonos issue rather than this repository

alphafox02, 2021-09-18:
Yep, sorry about that. Please clean up as needed. For DragonOS on the pi related questions/issues please go here.  

https://sourceforge.net/p/dragonos-pi64/tickets/

closed 2022-09-29 by AlexandreRouma