AlexandreRouma/SDRPlusPlus #341: SoapySDR DLL Doesn't Load on Windows 10

gstrike, 2021-09-07:
Hello,

I'm running SDR++ v1.0.3 on Windows 10.  I am attempting to utilize my hackrf, which is available on the network from an Ubuntu machine running SoapySDR / SoapySDRServer.

It appears the soapy_source.dll doesn't load when launching SDR++ with:

[error] Couldn't load ./modules/soapy_source.dll. Error code: 126

I'm not sure if this is a bug or something I'm missing.

[2021-09-07 11:38:15.822] [info] SDR++ v1.0.3
[2021-09-07 11:38:15.823] [info] Loading config
[2021-09-07 11:38:16.114] [info] Loading icons
[2021-09-07 11:38:16.130] [info] Loading band plans
[2021-09-07 11:38:16.132] [info] Loading band plans color table
[2021-09-07 11:38:16.136] [info] Loading modules
[2021-09-07 11:38:16.136] [info] Loading ./modules/airspyhf_source.dll
[2021-09-07 11:38:16.139] [info] Loading ./modules/airspy_source.dll
[2021-09-07 11:38:16.141] [info] Loading ./modules/audio_sink.dll
[2021-09-07 11:38:16.144] [info] Loading ./modules/bladerf_source.dll
[2021-09-07 11:38:16.146] [info] Loading ./modules/discord_integration.dll
[2021-09-07 11:38:16.147] [info] Loading ./modules/file_source.dll
[2021-09-07 11:38:16.149] [info] Loading ./modules/frequency_manager.dll
[2021-09-07 11:38:16.150] [info] Loading ./modules/hackrf_source.dll
[2021-09-07 11:38:16.152] [info] Loading ./modules/limesdr_source.dll
[2021-09-07 11:38:16.155] [info] Loading ./modules/meteor_demodulator.dll
[2021-09-07 11:38:16.158] [info] Loading ./modules/network_sink.dll
[2021-09-07 11:38:16.160] [info] Loading ./modules/plutosdr_source.dll
[2021-09-07 11:38:16.162] [info] Loading ./modules/radio.dll
[2021-09-07 11:38:16.163] [info] Loading ./modules/recorder.dll
[2021-09-07 11:38:16.165] [info] Loading ./modules/rigctl_server.dll
[2021-09-07 11:38:16.167] [info] Loading ./modules/rtl_sdr_source.dll
[2021-09-07 11:38:16.168] [info] Loading ./modules/rtl_tcp_source.dll
[2021-09-07 11:38:16.170] [info] Loading ./modules/sdrplay_source.dll
[2021-09-07 11:38:16.173] [info] Loading ./modules/soapy_source.dll
[2021-09-07 11:38:16.176] [error] Couldn't load ./modules/soapy_source.dll. Error code: 126
[2021-09-07 11:38:16.176] [info] Loading ./modules/spyserver_source.dll
[2021-09-07 11:38:16.177] [info] Initializing Airspy Source (airspy_source)
[2021-09-07 11:38:16.199] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-09-07 11:38:16.220] [info] Initializing Audio Sink (audio_sink)
[2021-09-07 11:38:16.220] [info] Initializing BladeRF Source (bladerf_source)
[2021-09-07 11:38:16.241] [error] Could not list devices
[2021-09-07 11:38:16.242] [info] Initializing File Source (file_source)
[2021-09-07 11:38:16.242] [info] Initializing Frequency Manager (frequency_manager)
[2021-09-07 11:38:16.243] [info] Initializing HackRF Source (hackrf_source)
[2021-09-07 11:38:16.264] [info] Initializing LimeSDR Source (limesdr_source)
[2021-09-07 11:38:16.265] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-09-07 11:38:16.266] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-09-07 11:38:16.284] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-09-07 11:38:16.285] [info] Initializing Radio (radio)
[2021-09-07 11:38:16.345] [info] RtAudio stream open
[2021-09-07 11:38:16.345] [info] Initializing Recorder (recorder)
[2021-09-07 11:38:16.346] [info] Initializing Rigctl Server (rigctl_server)
[2021-09-07 11:38:16.347] [info] Initializing SDRplay Source (sdrplay_source)
[2021-09-07 11:38:16.348] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2021-09-07 11:38:16.348] [info] Initializing SoapySDR Source (soapy_source)
[2021-09-07 11:38:16.348] [error] Module 'soapy_source' doesn't exist
[2021-09-07 11:38:16.348] [info] Initializing SpyServer Source (spyserver_source)
[2021-09-07 11:38:16.350] [info] Loading color maps
[2021-09-07 11:38:16.350] [info] Loading ./res/colormaps/classic.json
[2021-09-07 11:38:16.351] [info] Loading ./res/colormaps/electric.json
[2021-09-07 11:38:16.353] [info] Loading ./res/colormaps/gqrx.json
[2021-09-07 11:38:16.355] [info] Loading ./res/colormaps/greyscale.json
[2021-09-07 11:38:16.355] [info] Loading ./res/colormaps/inferno.json
[2021-09-07 11:38:16.356] [info] Loading ./res/colormaps/magma.json
[2021-09-07 11:38:16.357] [info] Loading ./res/colormaps/plasma.json
[2021-09-07 11:38:16.358] [info] Loading ./res/colormaps/turbo.json
[2021-09-07 11:38:16.359] [info] Loading ./res/colormaps/viridis.json
[2021-09-07 11:38:16.360] [info] Loading ./res/colormaps/websdr.json
[2021-09-07 11:38:16.376] [info] New DSP samplerate: 10000000.0 (source samplerate is 10000000.0)
[2021-09-07 11:38:16.377] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-09-07 11:38:16.378] [info] AirspySourceModule 'Airspy Source': Menu Select!
[2021-09-07 11:38:16.378] [info] New DSP samplerate: 10000000.0 (source samplerate is 10000000.0)
[2021-09-07 11:38:16.379] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-09-07 11:38:16.825] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:17.825] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:18.826] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:19.826] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:20.827] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:21.828] [warning] ConfigManager locked, waiting...
[2021-09-07 11:38:22.311] [info] RtAudio stream open
[2021-09-07 11:38:22.328] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-09-07 11:38:22.328] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-09-07 11:38:22.330] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-09-07 11:38:22.359] [info] AirspySourceModule 'Airspy Source': Tune: 100000000.0!
[2021-09-07 11:38:22.359] [info] Running post-init for Airspy Source
[2021-09-07 11:38:22.359] [info] Running post-init for AirspyHF+ Source
[2021-09-07 11:38:22.359] [info] Running post-init for Audio Sink
[2021-09-07 11:38:22.359] [info] Running post-init for BladeRF Source
[2021-09-07 11:38:22.359] [info] Running post-init for File Source
[2021-09-07 11:38:22.359] [info] Running post-init for Frequency Manager
[2021-09-07 11:38:22.359] [info] Running post-init for HackRF Source
[2021-09-07 11:38:22.359] [info] Running post-init for LimeSDR Source
[2021-09-07 11:38:22.359] [info] Running post-init for PlutoSDR Source
[2021-09-07 11:38:22.359] [info] Running post-init for RTL-SDR Source
[2021-09-07 11:38:22.359] [info] Running post-init for RTL-TCP Source
[2021-09-07 11:38:22.359] [info] Running post-init for Radio
[2021-09-07 11:38:22.359] [info] Running post-init for Recorder
[2021-09-07 11:38:22.360] [info] Running post-init for Rigctl Server
[2021-09-07 11:38:22.360] [info] Running post-init for SDRplay Source
[2021-09-07 11:38:22.360] [info] Running post-init for SpyServer Source
[2021-09-07 11:38:22.360] [info] Ready.

I've done some scouring to see if I'm missing something and found some references that made it look like PothosSDR may need to be installed, which I've done, but it didn't help.

AlexandreRouma, 2021-09-07:
You need to have SoapySDR installed on your system. It can't be shipped with SDR++.

Install PothosSDR and make sure to check the "Add to path" option in the installer

gstrike, 2021-09-07:
Thank you for the reply.  I did do that prior to my post (including the Add to Path)...

It is working now...  I needed to close/re-open the Command Prompt window...  Ugh.

Thanks again.

For those looking in the future, the latest Windows installer for SoapySDR can be found here: https://downloads.myriadrf.org/builds/PothosSDR/

closed 2021-09-07 by gstrike