AlexandreRouma/SDRPlusPlus #343: Changing sampling on RTL-SDR while streaming

cosmopotamian, 2021-09-07:
Would be quite handy to be able to change sampling frequency as streaming is going on. Currently, I have to stop it before the sampling frequency option becomes available.

AlexandreRouma, 2022-01-05:
No, the RTL-SDR can't change it's samplerate while the streaming samples.

closed 2022-01-05 by AlexandreRouma

shuyuan-liu, 2022-01-05:
How about stopping and restarting in the background so that the user doesn't have to stop and restart manually? I think this is very much doable, since @cosmopotamian didn't ask for seamless switching without losing samples or anything like that.

AlexandreRouma, 2022-01-05:
it'll likely cause issues with the DSP having to reconfigure to a new samplerate without all the samples being purged

shuyuan-liu, 2022-01-05:
I don't quite understand - if upon changing the sample rate SDR++ automatically stops DSP, makes the changes, and restarts DSP, wouldn't the effect be the same as doing it manually?

AlexandreRouma, 2022-01-05:
The reason is, forcing the user to stop, change an option, and then restart makes sure all the samples in the resamplers are consumed. If you instantly switch from a high samplerate to a low one and some data is still stuck before the resamplers, it'll have to be consomed when the DSP is restarted, causing non ideal behavior.

tl;dr, If I do it for the RTL, I have to do it for all sources, and rn I can't garantee this won't cause issues so i'd rather not take the risk for something that's not a big deal anyway