AlexandreRouma/SDRPlusPlus #345: Feature request: Soundcard I/Q input

batica81, 2021-09-07:
Maybe it is there but I cannot find it? It is a very useful feature for older SDRs and all kind of experiments.

AlexandreRouma, 2021-09-09:
Hello,

Currently there is indeed no audio source, in the mean time you can use the soapysdr source which supports soundcard input

batica81, 2021-09-09:
Good to know, I will try that, thanks.

jyrts, 2022-01-30:
with soapysdr sound input there are two problems:
1. You can not specify channel mode. How to use IQ audio input?
2. If you have multiple audio devices for different bands on you soapy remote server then you can not specify wich soundcard to use. How to specify device_id?

AlexandreRouma, 2022-09-30:
@bubnikv I appreciate the gesture but there is no need, the audio source can be implemented using anything as input. Control of the frequency will be done through the rigctl client.

AlexandreRouma, 2023-02-21:
Implemented

closed 2023-02-21 by AlexandreRouma

batica81, 2023-02-21:
Works well with QDX radio controlled by flrig, also with any old SDR connected to Asus USB soundcard! Thank you!

arampak, 2023-08-25:
> Implemented

I am unable to find a way to use the radio that has an IQ output connected to the USB sound card input in on my Mac. What are the particular steps to make SDR++ recieve the IQ data from a sound card?