AlexandreRouma/SDRPlusPlus #350: Ham it up miscalculation

wrenchtamer, 2021-09-10:
I just noticed that there is a bug in the new release for Linux that incorrectly calculates the offset when using the ham it up up converter.  It functions correctly when you enter the offset manually.  I'm running DragonOS focal.  Sorry for any lack of understanding in terms of posting guidelines.

AlexandreRouma, 2021-09-10:
can you elaborate on your issue? The ham-it-up offset is set to 125MHz, this is what the docs of the hamitup say it should be.

wrenchtamer, 2021-09-10:
It should be -125mhz or -125000khz. The offset should be negative instead of positive.

AlexandreRouma, 2021-09-10:
that's not how it works in sdr++. The offset is positive.
And it's indeed set to 125MHz

wrenchtamer, 2021-09-10:
If you say so.  I'll check again when I get a chance.  Have you actually tested it with the hardware?

JosephTheTank13, 2021-10-05:
SDR++ puts the offset at 125000000.000Mhz where when you put it in manually you set it to 125.000. The issues is with the decimal point when choosing Ham-It-Up in SDR++. 125.000 is the same as 125Mhz but what SDR++ is doing is up converting it by 125,000,000Mhz.

AlexandreRouma, 2021-10-05:
That's not true. The manual entry is in Hz with a dot as the decimal place.

JosephTheTank13, 2021-10-05:
I tested it now with a Ham-It-Up-Plus and it appears to be working properly. It used to be messed up though, I guess it was fixed at some point.

AlexandreRouma, 2021-10-05:
Haven't touched that part of the code in a few months so doubt it. You probly just entered something wrong the first time.
I use the offset almost every day for upconverters or to use my SDRs as panadaoters for my rig, never had any issue with the offset being wrong

JosephTheTank13, 2021-10-05:
I can confirm it was bugged a while ago though... something must have changed.