AlexandreRouma/SDRPlusPlus #353: RTL-SDR needs Zadig for Windows SDR++ to work?

jejesson, 2021-09-12:
I have other SDR's work perfectly (Airspy, etc.), just garden-variety RTL-SDR does not seem to connect and run despite the "refresh" button sees RTL-SDR. I tried deleting the default Windows drivers but that does not help.

Best, Joe, W2JEJ

shuyuan-liu, 2021-09-12:
You do need to install drivers for the RTL-SDR with Zadig. In my understanding this is needed regardless of which SDR software you use.

lilkrucivert, 2022-02-27:
how about manjaro? i don't have a RTL-SDR option in "source" menu

shuyuan-liu, 2022-02-27:
@lilkrucivert you need to install the rtl-sdr package.

lilkrucivert, 2022-02-27:
I don't have the option to add anything in Module Manager tab. I can only remove things. I have deleted my config at least twice and I still don't have RTL-SDR option i Source menu

closed 2022-09-13 by AlexandreRouma