AlexandreRouma/SDRPlusPlus #354: Latest Windows Build Binaries not running on OpenGL 2.1 Card

Voyager-III, 2021-09-13:
Hi, I tried the latest Windows Build Binary on my laptop with an OpenGL 2.1 Card it tries to start with a white screen and crashes? but when I compile the linux version from the master branch it runs perfect, I see that you have added OpenGL 2.1 support but does it only run on linux for the moment, Thanks

AlexandreRouma, 2021-09-13:
Can you run SDR++ from a terminal with the `-s` command line argument and send back what it logs?

Voyager-III, 2021-09-13:
Hi, Thanks for getting back to me this is the log

[2021-09-13 21:02:19.095] [info] SDR++ v1.0.4
[2021-09-13 21:02:19.097] [info] Loading config
[2021-09-13 21:02:19.358] [error] Glfw Error 65543: Requested OpenGL version 3.0
, got version 2.1
[2021-09-13 21:02:19.361] [info] OpenGL 3.0 was not supported
[2021-09-13 21:02:19.371] [error] Glfw Error 65542: WGL: OpenGL ES requested but
 WGL_ARB_create_context_es2_profile is unavailable
[2021-09-13 21:02:19.372] [info] OpenGL 3.1 ES was not supported
[2021-09-13 21:02:19.486] [info] Using OpenGL 2.1

and then it start's to load a white screen and crashes, my graphics card is a GeForce Go 7400 which supports OpenGL 2.1

AlexandreRouma, 2021-09-13:
is there really nothing else printed in the console after that?
I need the full thing to know what's really going on. I doubt this has anything to do with opengl

Voyager-III, 2021-09-13:
I just return to the command prompt, I will try to upload a screenshot
[image]

AlexandreRouma, 2021-09-17:
Please show a screenshot of the terminal when it crashes

Voyager-III, 2021-09-17:
Hi, Here is the screenshot of the command line when it crashes and another when I close the progra
[image]
[image]
m

AlexandreRouma, 2021-09-17:
Ok, are you sure SDR++ has read/write file access rights where you decompressed it? 

Also, make sure your graphics drivers are up to date.

I'd also be interested in the specs of the computer (CPU/GPU/RAM)

Voyager-III, 2021-09-17:
This is my screenshot of my CPU and GPU I have 4.00GB RAM (3.00GB) Usable 64bit Windows 8.1 I have tried Windows 7 and it has the same problem, The graphics driver is the latest possible the built in card is not supported anymore
[image]

AlexandreRouma, 2021-09-17:
How much free ram have you *actually* got when starting SDR++?
Windows is pretty well known for hoarding a bunch of ram at startup...
SDR++ needs almost 2GB last time I checked, mostly because of OpenGL

Voyager-III, 2021-09-17:
This is how much ram I have left it seem SDR++ does not use much when trying to start up but Windows Defender eats through it, It is just the same even when I turn off windows defender
[image]

266-750Balloons, 2021-09-23:
I'm pretty sure I had the same problem trying to get SDR++ running on an old laptop on Linux today. The issue has been resolved in 1.04 (see #260), but for some reason, there are no binaries for 1.04. I had to build it from source. After many hurdles, I got it running on a 2010 laptop like a charm. You should ask the developer to put out binary releases with the fix, which came out 12 days after the most recent binary.

266-750Balloons, 2021-09-23:
The 1.03 binary doesn't work on Linux for same reason. I had to build 1.04 from source.

AlexandreRouma, 2021-09-23:
If you look at the logs, he's already running 1.0.4
Also, the latest commit is always available in the automated builds.

gvanem, 2021-10-02:
> SDR++ needs almost 2GB last time I checked, mostly because of OpenGL and the enormous waterfall history buffer

Depends on the FFT-size, doesn't it? I had some crashes in `waterfall.cpp` with 64k. But going down to 32768, it's rather stable.

Several places in that code is not checking for `rawFFTs == NULL` followed by a `memset(rawFFTs, 0, ...);`.

AlexandreRouma, 2021-10-02:
because those places are never called when rawFFTs is null. waterfall.init() gets called way before that.
I've checked several times with valgrind and asan, it doesn't have any kind of issue

gvanem, 2021-10-02:
Really? When e.g. toggling off the `"Display | Show Waterfall"` check-button, and turning it on again, I once got this crash:
```
eax=00000000 ebx=009df8c4 ecx=04540000 edx=00f8a7a0 esi=009dea94 edi=00000000
eip=69bd381c esp=009de538 ebp=009de54c iopl=0         nv up ei pl nz na po cy
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00010203
VCRUNTIME140!memset+0x3c:
69bd381c f3aa            rep stos byte ptr es:[edi]
```
The Call-stack clearly shows a `memset()`  on a NULL-ptr (`edi == 0` above and `count` == 10 349 204 bytes below):
```
VCRUNTIME140!memset(unsigned char * dst = 0x009dea94 "", unsigned char value = 0xa0 '', unsigned long count = 0x9dea94)+0x3c
sdrpp_core!ImGui::WaterFall::showWaterfall(void)+0x4f
sdrpp_core!displaymenu::draw(void * ctx = 0x00000000)+0x9f
sdrpp_core!Menu::draw(bool updateStates = false)+0x860
sdrpp_core!MainWindow::draw(void)+0x1641
sdrpp_core!sdrpp_main(int argc = 2, char ** argv = 0x00f81b28)+0x36f9
sdrpp!main(int argc = 2, char ** argv = 0x00f81b28)+0x10
sdrpp!invoke_main+0x1c
sdrpp!__scrt_common_main_seh(void)+0xfa
KERNEL32!BaseThreadInitThunk+0x19
ntdll!__RtlUserThreadStart+0x2f
ntdll!_RtlUserThreadStart+0x1b
```

Unsure about the `value = 0xa0`. This on a  32-bit version BTW.

AlexandreRouma, 2021-10-02:
The only way it can be null is if a malloc operation failed. (try searching for `rawFFTs = ` and look at the only code assigning it)
I don't test on 32bit but I'm 90% sure it's the same thing and it's running out of memory. OpenGL takes a lot of memory by itself and when you add all the buffers for the DSP + the waterfall it's probably too much for 32bit windows to handle....

app7713-design, 2021-10-02:
hi prep myself id try vars options with your graphics card using linux then
access denied opengl 3.0-1 because in some cases there has been issues were
i found myself unable to use open gl software graphics in windows but yet
run smootlys in others alex your doing a great job mate

gvanem, 2021-10-02:
> (try searching for rawFFTs = and look at the only code assigning it)

I know. I put some trace-code around those places; always `realloc()` that fails.
Could such a huge buffer be pre-allocated at start-up instead? (based on max FFT-size etc.)

AlexandreRouma, 2021-10-02:
That would be a complete was of memory. The point alloc reallocating is to use less ram.
If malloc works but realloc fails, it's a bug in the libc and I can't do much about it.

AlexandreRouma, 2022-09-30:
in retrospect it was probably a driver issue knowing how trash anything made by nvidia is, I'll close for now.

closed 2022-09-30 by AlexandreRouma