AlexandreRouma/SDRPlusPlus #358: SDRPlay Issue

Charlier26, 2021-09-13:
I have installed v1.03 on Windows 10-64bit, and while when I select SDRPlay as the radio, it recognizes my RSP1A, and lists its serial number, below that in red it says "No device available". What am I doing wrong. If I try to start the radio, it shows no signals, and then no longer recognizes the receiver.

AlexandreRouma, 2021-09-13:
Make sure no other software is using your RSP1A. A reboot can help.

Charlier26, 2021-09-13:
I don't have any other program running, but DO have SDRPlay's API server running....could that be conflicting with it?

AlexandreRouma, 2021-09-13:
If by the API server you mean the API service, it has to be running for SDR++ to work.
Only advice I can give to to uplug/replug the device and reboot

Charlier26, 2021-09-13:
OK, I'll try that

Charlier26, 2021-09-13:
I reloaded the API service, and I got it running......but it sure is deaf. In SDRUno, I get plenty of signals.  The RF Gain is all the way up, and the IF gain isn't available to adjust but it looks like it is at max.

AlexandreRouma, 2022-09-13:
either user error or early bug, try the nightly

closed 2022-09-13 by AlexandreRouma