AlexandreRouma/SDRPlusPlus #360: [Feature Request] DSD support

AsciiWolf, 2021-09-15:
Hello and thanks for this great new SDR application! I think that it would be great if SDR++ had direct support for Digital Speech Decoder, to decode DMR, D-STAR and other digital modes, just like SDR# does. :-)

AlexandreRouma, 2021-09-16:
Hello,
This is a planned feature but I don't expect it to be ready before the beginning of next year

anon88391, 2022-03-16:
@AlexandreRouma How is the progress on this?
This is the only feature that prevents me from stopping using SDR# and I'm really anticipating it.

AlexandreRouma, 2022-09-30:
coming back to it, "DSD" is extremely vague. Support for some digital modes will be added with time (M17 is already supported), but I can't add every possible digital voice mode. Moreover I will not integrate any other DSD application in SDR++. It all must be written from scratch to ensure code quality and maintainability.

In conclusion I'm closing this issue and I recommend specific modes be disussed in other issues instead.

closed 2022-09-30 by AlexandreRouma