AlexandreRouma/SDRPlusPlus #361: BIAS TEE SEEMS TO BE NOT WORKING IN RTL-SDR?

sefirot43, 2021-09-16:
Good morning Alexandre, maybe I am doing something wrong, but it looks that the bias tee option in rtl-sdr (usb) seems to be not working properly.

AlexandreRouma, 2021-09-16:
Hello, 

please make sure that your RTL-SDR supports bias-t.

chriswei, 2021-09-23:
I just verified that it seems to work on my setup. Mac OSX 11.6, "official" RTL-SDR.com blog V3 dongle, SDR++ 1.0.4, Airspy Spyverter2 and MLA-30+ loop antenna.

I tested SDR++'s Bias T setting on both my Spyverter v2 upconverter and directly to my MLA-30+ antenna (without the Spyverter in the chain), and it seemed to work for both: Bias T checked, clear signals received. Bias T unchecked, no signals, just noise.

I built SDR++ following Ryzerth's Discord post instructions (https://discord.com/channels/752669522883903601/752672743690076284/874740527839453325) and a couple loose ends with glfw and a missing libairspyhf.

AlexandreRouma, 2021-09-24:
the RTL-SDRblog V3 is to my knowledge the only rtlsdr that has a selectable bias-T