AlexandreRouma/SDRPlusPlus #364: Not starting up

k9hacker, 2021-09-18:
Ran the following:

```
sudo apt install libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev
sudo dpkg -i sdrpp_ubuntu_hirsute_amd64.deb
```

Start up, but this happens and doesn't startup until the system asks to force quit or wait.

[image]

k9hacker, 2021-09-18:
```
NAME="Pop!_OS"
VERSION="21.04"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.04"
VERSION_ID="21.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=hirsute
UBUNTU_CODENAME=hirsute
LOGO=distributor-logo-pop-os
```

shuyuan-liu, 2021-09-18:
Looks the same as #192. See https://github.com/AlexandreRouma/SDRPlusPlus#hash-collision-error-when-starting for solution.

AlexandreRouma, 2021-09-18:
Indeed looks like it

closed 2022-09-29 by AlexandreRouma