AlexandreRouma/SDRPlusPlus #367: 2 issues may be easy fixes

RjulianTN, 2021-09-20:
Thanks for a great program!  2 things stopping me.  First I see my SDR dongle and I see it getting and displaying signals
The volume control is locked to mute and zero volume.  Nothing I do will change it.
Second issue may fix the first.  What do i need to blow away to make a reinstall total?  Removing the packages leaves some type of config files or something behind as there are modules missing upon reinstall.

I'm using a generic RTL-SDR dongle.
Ubuntu Mate latest stable build

Thanks again
Ronny 
K4RJJ
[image]

shuyuan-liu, 2021-09-20:
First issue: Make sure in the Sinks tab you select an option other than "None".

Second issue: not very sure where the files are located, will have a look when I can.

RjulianTN, 2021-09-20:
Interesting.  My Sinks tab is there but there is nothing to select.  I can open/close it (arrow down or right) but there are no controls below that tab.  I also do not have a mode selection tab am,fm,ssb etc..

AlexandreRouma, 2021-09-20:
I assume this is an issue with a broken config since having audio as a sink and a default VFO is the normal default config. 
The config files are located in `~/.config/sdrpp`. I'd suggest deleting this directory and trying again.

By the way, make sure rtaudio is installed or you won't have any audio output.

RjulianTN, 2021-09-20:
This was EXACTLY the issue.  Don't know how it was corrupted.  Is there a way this can be an option in the program later?  Getting rid of this directory and starting fresh works well!  Thanks very much!  
Ronny K4RJ
[image]
J

RjulianTN, 2021-09-20:
BTW RT Audio was installed correctly but I checked it anyway.

closed 2021-09-20 by RjulianTN