AlexandreRouma/SDRPlusPlus #372: Ubuntu 20.04 and stubborn request for LimeSDR

Sinager, 2021-09-22:
Hi,
I'm unable to run the "release" packaged SDRPP.

This is the error I get:
[2021-09-22 09:52:44.762] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2021-09-22 09:52:44.777] [error] Couldn't load /usr/lib/sdrpp/plugins/limesdr_source.so.
...
[2021-09-22 09:52:45.010] [info] Initializing Radio (radio)
malloc(): unsorted double linked list corrupted

A similar issue with BladeRF was cured by simply removing the entry in config.json file, as I don't care for these two receivers, the same doesn't work for Lime.

Similar bug was already reported and closed when the author was able to solve by removing soapysdr packages, but not working for me.

File is there
-rw-r--r-- 1 root root  536376 ago 10 01:00 limesdr_source.so

Thank you

AlexandreRouma, 2021-09-22:
Hello,
I do't think the issue has anything to do with some of the modules not being able to be loaded, SDR++ just prints an error and moves on.

What's more worrying however is the segfault on the radio module. I think this might be related to #321

AlexandreRouma, 2022-09-29:
like has been fixed since then, reopen if not