AlexandreRouma/SDRPlusPlus #373: MacOS 10.12.6 Sierra: fftw issue

benb0jangles, 2021-09-22:
Installed using sdr++ dmg:
https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.3/sdrpp_macos_amd64.pkg

ran in terminal:
sdrpp

gives error:
```
dyld: Library not loaded: /usr/local/opt/fftw/lib/libfftw3f.3.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: Incompatible library version: sdrpp requires version 10.0.0 or later, but libfftw3f.3.dylib provides version 3.6.9
Abort trap: 6
```

attempted to fix using:
`brew install fftw`

gives error:
`Error: You are using macOS 10.12.`

it seems MacOS 10.12.6 Sierra uses xcode v9
Apple LLVM version 9.0.0 (clang-900.0.39.2)

and is unable to install/upgrade to fftw 3.3.10 via Homebrew due to older xcode limitation.

I am able to install fftw using macports:
`sudo port install fftw-3`
however it only installs older version of fftw
trying to upgrade using:
`sudo port selfupdate && sudo port upgrade fftw-3`
does not upgrade fftw.

Any ideas on a solution or workaround?

AlexandreRouma, 2021-09-22:
Hello,
SDR++ is currently only compatible with MacOS Catalina and above.

benb0jangles, 2021-09-22:
ok thanks.

closed 2021-09-22 by benb0jangles