AlexandreRouma/SDRPlusPlus #374: No UI scaling feature?

jaredkaragen, 2021-09-22:
I use a 4K monitor, with windows scaling for apps to not be tiny.

It seems SDR++ is still showing as if my desktop UI scaling was off;  so I would request some sort of UI scale factor that you can adjust would be an excellent ad. 

I simply can not read anything from the app unless I change my desktop resolution or get up against the screen.

Not exactly a bug, but more of a feature request for creature comfort.

AlexandreRouma, 2021-09-22:
Scaling isn't currently supported but it's a planned feature

steve-m, 2022-02-12:
Same issue here, I can barely read the text on my 4K screen.

AlexandreRouma, 2022-02-12:
it's being worked on at the moment since it was required to make the thing usable on android

AlexandreRouma, 2022-09-13:
UI scaling was added and is now available

closed 2022-09-13 by AlexandreRouma