AlexandreRouma/SDRPlusPlus #377: Any planned support for VNC

maldea, 2021-09-23:
I've tried installing SDR++ 1.0.3 on Windows 10 over VNC. The SDR++ app won't start. Trying to launch it from command prompt gets me nowhere either:

```
c:\Users\[redacted]\Desktop\SDRPP>sdrpp.exe
[2021-09-23 17:39:11.656] [info] SDR++ v1.0.3

c:\Users\[redacted]\Desktop\SDRPP>sdrpp.exe -s
[2021-09-23 17:39:14.566] [info] SDR++ v1.0.3
[2021-09-23 17:39:14.568] [info] Loading config
[2021-09-23 17:39:14.592] [error] Glfw Error 65542: WGL: The driver does not appear to support OpenGL
```

Same happens on another machine using RDP. 

Issue #234 brings this up to but there's no solution.

Are you considering adding support for such scenarios where SDR++ runs on a remote machine, eg. for FT-8 receive only?

AlexandreRouma, 2021-09-23:
this is not a VNC issue, it's an OpenGL issue, make sure your graphics drivers are up to date.
Not using OpenGL would negate all the performance improvements that using graphics acceleration procurs. Any PC built in the last 10 years should have at least OpenGL 2.1 and run SDR++ just fine.

maldea, 2021-09-23:
You're absolutely right. I haven't bothered to use anything other that the Windows 10 embedded drivers on my Intel NUC since I always use it over VNC or RDP.
After installing the official Intel Graphics card drivers the app fired up with no issue. And what a wonderful UI it has, brings back memories from years ago when SDR# was clean and easy to use. Good thing you took over. Thanks a lot.

AlexandreRouma, 2021-09-23:
Good to know!
btw, I didn't really "take over" SDR#, this is completely new software from scratch ;)

closed 2021-09-23 by AlexandreRouma