AlexandreRouma/SDRPlusPlus #378: M17 mode decoder

sp5wwp, 2021-09-24:
It'd be really nice to have M17 decoder in your software.

In case you haven't heard about it already, it's a fully open-source digital communication mode made by hams.
Our website is m17project.org, Twitter: twitter.com/m17_project

AlexandreRouma, 2021-09-24:
Hello,
If the phy and codec layers are relatively simple it should be a quick addition.

Could you provide a few IQ recordings of the protocol (if possible of different SNR and containing most features of the protcol)?
I don't think I own hardware that can use that mode

sp5wwp, 2021-09-25:
If you have a Raspberry Pi (even Zero) you can use [rpitx by F5OEO](https://github.com/F5OEO/rpitx) to transmit M17. We can provide a baseband sample for you to experiment.

[Contact us directly](https://m17project.org/contact/) for more details.

sp5wwp, 2021-09-25:
There's [coder/decoder ready](https://github.com/mobilinkd/m17-cxx-demod), written in C++. Take a look :-)
Also, check the [specification](https://m17-protocol-specification.readthedocs.io/en/latest/), it's pretty straightforward.

AlexandreRouma, 2021-09-25:
Alright I'll give it a try with RPi TX, hopefully my currect clock recovery algo (M&M) is good enough for packet operation.
I've joined the matrix and the discord

AlexandreRouma, 2021-10-19:
Added in this release

closed 2021-10-19 by AlexandreRouma