AlexandreRouma/SDRPlusPlus #381: Native Apple Silicon Support

bvogt3000, 2021-09-24:
First, thanks for the new SDR program! I am a mac guy and love SDR radio, so it is nice to finally see a new SDR program come to the Mac. Right now SDR++ is running through Rosetta on my m1 Mac Mini, it works rather well. I am still wondering though if there is a plan to have native Apple Silicon support?

AlexandreRouma, 2021-09-24:
Hello,
Unfortunately don't own (and can't afford) any mac so I won't be able to add support for apple silicon myself.

bvogt3000, 2021-09-24:
Ok, I just figured I would check. Nonetheless thanks for the SDR program! It still runs great through Rosetta.

l-gothberg, 2021-10-03:
> Hello, Unfortunately I don't own (and can't afford) any mac so I won't be able to add support for apple silicon myself.

What if you could?

AlexandreRouma, 2021-10-03:
> What if you could?

Unless \<insert price of a M1 machine here\> magically appears in my bank account, I can't
I'm just a student, I don't have the cash nor the need for a brand new apple device
Even used they're almost MSRP anyway...

Tried emulation for intel macs and that kinda works but I can't actually start the software since it needs OpenGL...

l-gothberg, 2021-10-03:
> Unless <insert price of a M1 machine here> magically appears in my bank account, I can't I'm just a student, I don't have the cash nor the need for a brand new apple device Even used they're almost [...]

Check your email.

AlexandreRouma, 2021-10-03:
Checked and responded 👍

n2iw, 2021-10-12:
> First, thanks for the new SDR program! I am a mac guy and love SDR radio, so it is nice to finally see a new SDR program come to the Mac. Right now SDR++ is running through Rosetta on my m1 Mac [...]

If you know how to compile, you can compile SDR++ from source. I compiled it on my M1 mini, and it runs natively. When I got sometime, I'll try to learn how to package Mac Applications and help provide ready to use application. I'm a software engineer, but I write web applications, C/C++ is not my strength.

AlexandreRouma, 2022-09-29:
Apple silicon is supported but sitll haven't got hardware. no reason for it to not work so I'll close.

closed 2022-09-29 by AlexandreRouma