AlexandreRouma/SDRPlusPlus #383: macos using /usr/local but all my libs etc are in /opt/local

pidloop, 2021-09-24:
Hello! Many thanks for SDR++ and for supporting macOS. Unfortunately, `otool -L sdrpp` reports many libs are required in /usr/local but on my system they are all in /opt/local. I suppose I could make a zillion symlinks but before I try that I'm wondering if there is a better way.

AlexandreRouma, 2021-09-24:
Hello,
Please install the library using homebrew.

pidloop, 2021-09-25:
Ok, I use macports, so that explains it. I find mixing the two can be problematic so not sure what I'll do.

Would be nice if you just packaged the libs with the kit. Gives you a stable base for testing too. I'm also a developer and I've been busted by open source libs changing and making my apps look at fault.

Anyway, many thanks for thinking of macOS at all. beep boop. 73.

closed 2021-09-25 by pidloop