AlexandreRouma/SDRPlusPlus #386: Segmentation fault with Radioberry Soapy

madpsy, 2021-09-25:
I am getting a seg fault as follows when starting the application on my Pi4. The last reference relates to Radiobery (an SDR Pi Hat).

Given I don't actually need soapy support as I just want to use it with an RTL-SDR is the quickest workaround to complile without Soapy support? If so, what's the build flag to not compile in Soapy? Thanks in advance.

`[2021-09-25 14:05:29.761] [info] Initializing SoapySDR Source (soapy_source)
[INFO] SoapyRadioberry::SoapyRadioberry  constructor called
[INFO] SoapyRadioberry::listAntennas called
[INFO] SoapyRadioberry::listGains called
[INFO] SoapyRadioberry::listBandwidths called
[INFO] SoapyRadioberry::SoapyRadioberry  destructor called
Segmentation fault
`

madpsy, 2021-09-25:
I just tried moving `/usr/lib/sdrpp/plugins/soapy_source.so` and the software now starts, albeit crashes when I hit play after selecting RTL-SDR but that may be a different issue.

AlexandreRouma, 2021-09-25:
Make sure the DVB-T drivers are blacklisted and the ones for the RTL-SDR are loaded instead. (details on the rtl-sdr repo)

madpsy, 2021-09-25:
thanks for replying- I have that blacklisted (and indeed it works with GQRX) but seems the issue is that X crashes rather than sdr++ so that's a whole other rabbit hole for me to debug!