AlexandreRouma/SDRPlusPlus #390: Audio is choppy.

BloodBlight, 2021-09-27:
When starting audio it is choppy, if I switch to a different audio output it clears up.  It continues to be fine if I switch back.

This happens when ever I stop/start the audio.  And sometimes will start happening again while just listening.

Switching channels or output frequency doesn't seem to have any impact.

Short video showing the issue with the sound monitor open (and flickering):
https://www.youtube.com/watch?v=jlRxuI0IUNQ

A longer video showing showing the behavior when switched:
https://youtu.be/GuxCpGXjwp8

Currently running:
Linux Mint 20 Ulyana
Ryzen 3900X
64GBs or RAM

AlexandreRouma, 2021-09-27:
This is a common bug on debian based (and especially mint) systems, follow the instructions outlined here to fix the issue: https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

BloodBlight, 2021-09-27:
Hmm, it looks like my rtaudio is librtaudio6, and there are a lot of dependencies on it including soapy and shotcut:
```
The following packages will be REMOVED:
  cubicsdr kdenlive libmlt++3 libmlt6 librtaudio6 melt shotcut soapysdr-module-all
  soapysdr0.7-module-all soapysdr0.7-module-audio
```

Can this be installed without removing the module?

BloodBlight, 2021-09-27:
Maybe this calls for an appimage?

AlexandreRouma, 2021-09-27:
No, just install over it.

BloodBlight, 2021-09-27:
Yep, that worked!

closed 2021-09-29 by AlexandreRouma