AlexandreRouma/SDRPlusPlus #391: How reset configuration following crash?

K0HYD, 2021-09-29:
I was running fine with airspy on 1.03 and with both airspy and airspyhf modules with1.04. Then I tried to use spysever. Since I set it as a source sdrpp crashes on launch and I don't know how to get rid of the spyserver setting so I can use sdrpp again. When I tried to configure spyserver I couldn't type in the ip address field. I like this program and signed up with Pateron. Any suggestion appreciated.

K0HYD, 2021-09-29:
MacOS builds. Running Monterey (MacOS 12, Beta 7). Forgot to note this in initial message.

AlexandreRouma, 2021-09-29:
Hello,
Could you send startup logs to see when it crashes?
You can reset the configuration by deleting `~/.config/sdrpp` but don't do it now, I'd like to know why it crashes and fix the issue.
Would be nice if you could zip up that folder and send it here so I can have a look at the config files causing the issues

K0HYD, 2021-09-29:
[ConfigArchive.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7249613/ConfigArchive.zip)
[TroubleReportLog.rtf.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/7249670/TroubleReportLog.rtf.zip)

K0HYD, 2021-09-29:
Hi!

The files in the .config folder and the TroubleReport log that goes to apple are attached. Hope this is the log you were looking for. Probably has a lot more information than you needed but hope it helps.

K0HYD, 2021-09-29:
Sorry didn't mean to close the issue. Let me know if you can't get the files. 73 de Dale

AlexandreRouma, 2021-09-29:
Alright thanks for this. I'll have a look at the code, the place where it crashes it pretty worrying.

AlexandreRouma, 2021-09-29:
Just wondering, did resetting the config actually fix it?

K0HYD, 2021-09-29:
Yes, I just deleted the entire directory and fired up sdrpp again and it came on just fine pointed to my airspy device. I may wait awhile to try the spyserver again. Thanks for the quick response.

Dale

AlexandreRouma, 2022-09-30:
these issues are likely fixed by now, reopen if not

closed 2022-09-30 by AlexandreRouma