AlexandreRouma/SDRPlusPlus #392: FM and SSB bandwith too small

nbserver, 2021-09-29:
Hi,
Great project!
I am using Your APP to satellite monitoring I am having some problems with decoding signals.
The problem is that the bandwith is too narrow for USB modulation, the old sdr# versions had 16kHz, the newer ones don’t have any limitations in FM and SSB. I would like to ask is it possible to increase this width for FM and SSB in SDR ++?
Regards!

AlexandreRouma, 2021-09-29:
If you need higher bandwidth on FM you can use WFM up to 250Khz, though i'm not quire sure why you would need that.
For APT satellites I use 44KHz in NFM.

In SSB, the bandwidth is limited because of the sample rate at which the demodulator runs. I didn't see the need for over 12KHz in USB, if you need to export the IQ for another decoder use the RAW mode instead.

nbserver, 2021-09-29:
Thanx for reply,
in a case of ssb some new satelitties working with psk over 14kHz. I mean live pass, not iq recordings. Live decoding is not possible with 12kHz so is there any chance to not limit or broad the bandwith?

AlexandreRouma, 2021-09-29:
It's not possible to increase the maximum bandwidth without wasting CPU for everyone using it at more normal bandwidths because that would require increasing the sample rate of the demodulator. If you want to live demod, use the RAW mode and use a decoder that can take IQ as input instead of sound

nbserver, 2021-09-29:
OK, thanx for reply.

closed 2021-09-29 by nbserver