AlexandreRouma/SDRPlusPlus #394: Build/run issue with in Gentoo.  "Config is corrupted, resetting it"

clytle374, 2021-09-29:
This is a build from todays git source. I'm on Gentoo, so a source build is my only real option.  "Running for development"  is what I'm shooting for so I don't mess the system up.  Target is a SDRplay RSP1A

FYI: I shut off the following airspy, airspyHF, bladeRF, hackRF, limesSDR, plutoSDR to get the software to configure.
Also had to shut off SDRplay due to a build failure.  
`/home/cory/sources/SDRPlusPlus/sdrplay_source/src/main.cpp:10:10: fatal error: sdrplay_api.h: No such file or directory
   10 | #include <sdrplay_api.h>
`
Even though I have sdrplay 2.13.1 through the system build manager, just skipped it to see if I could get the software up and running.  This is just background in case it matters.

`sources/SDRPlusPlus/build $ find . | grep '\.so' | sed 's/^/"/' | sed 's/$/",/' | sed '/sdrpp_core.so/d'
"./audio_sink/audio_sink.so",
"./soapy_source/soapy_source.so",
"./network_sink/network_sink.so",
"./file_source/file_source.so",
"./meteor_demodulator/meteor_demodulator.so",
"./recorder/recorder.so",
"./frequency_manager/frequency_manager.so",
"./weather_sat_decoder/weather_sat_decoder.so",
"./radio/radio.so",
"./discord_integration/discord_integration.so",
`
If I input that into config.json like so. 
`"modules": [

"./audio_sink/audio_sink.so",
"./soapy_source/soapy_source.so",
"./network_sink/network_sink.so",
"./file_source/file_source.so",
"./meteor_demodulator/meteor_demodulator.so",
"./recorder/recorder.so",
"./frequency_manager/frequency_manager.so",
"./weather_sat_decoder/weather_sat_decoder.so",
"./radio/radio.so",
"./discord_integration/discord_integration.so",
]
`
I get this error.  tried several different combinations of tabs, and commas at the end of lines, but always get this.
`[2021-09-29 11:40:25.156] [info] SDR++ v1.0.4
[2021-09-29 11:40:25.156] [info] Loading config
[2021-09-29 11:40:25.156] [error] Config file 'root_dev/config.json' is corrupted, resetting it
[2021-09-29 11:40:25.157] [error] Resource directory doesn't exist! Please make sure that you've configured it correctly in config.json (check readme for details)
`
If I leave the modules part out and just set the resource and module directories, the software will start. 
I'm using nano as the editor.  Taken like 50 shots at the file, and getting exhausted with it.  Deleting it seems like cruel punishment of a mistake I don't understand.  I should be able to provide any information needed. 

Thanks
Cory
Any ideas on the sdrplay failure would be great.

AlexandreRouma, 2021-09-29:
I'd really recommend just installing it but if you really want to run for development (which should never be done for normal use because it forces you to start with command line arguments and from a specific directory), you need to properly configure the path to the resource directory as indicated in the readme. (as well as to the module directory even if it's empty).

clytle374, 2021-09-29:
> I'd really recommend just installing it but if you really want to run for development (which should never be done for normal use because it forces you to start with command line arguments and from a [...]

Installing it would be better. I build and run in place many programs without issue. The one I do have to install causes weird issues with other parts of the build system. 

I realize it's probably going to take a start script.  But following the readme on this page really doesn't work, hence the question.
Thanks
Cory

clytle374, 2021-09-29:
Alright, installed it, new thread

closed 2021-09-29 by clytle374