AlexandreRouma/SDRPlusPlus #395: Build/install issues with sdrplay and libsdrpp_core.so Gentoo Linux from git

clytle374, 2021-09-30:
Per the advice here I installed the software instead of trying to run it in place.  I run freecad, prusaslicer,  and a few others from source and run in place.  I also run kicad from source, but it has to be installed and causes issues.  
First issue is I get error on startup.

`sdrpp: error while loading shared libraries: libsdrpp_core.so: cannot open shared object file: No such file or directory`

I found libsdrpp_core.so in /usr/lib instead of /usr/lib64 so copied it over to lib64 and the program starts.  Gentoo strikes again.  Looking at cmake-gui didn't give me any ideas how to fix it properly.

I have installed 
soapysdr 0.7.2
soapysdrplay 0.2.0
sdrplay 2.13.1
Which allows gqrx to run and talk to the sdrplay RSP1A on this system.  But not SDRPlusPlus.

When trying to build with sdrplay option I get the following build failure.
`/home/cory/sources/SDRPlusPlus/sdrplay_source/src/main.cpp:10:10: fatal error: sdrplay_api.h: No such file or directory
   10 | #include <sdrplay_api.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
`
Is this due to API 2 being installed instead of API 3?

Any help would be appreciated.

AlexandreRouma, 2021-09-30:
You should symlink it, not copy. Because if you rebuild and install a lter version you'll have to copy it every time.
As for SDRplay, you need the v3 API

clytle374, 2021-09-30:
Yes, symlink is the correct answer.  Annoying it won't run in place after following the instructions. 
Okay, the new API answers the other question.

closed 2021-09-30 by clytle374

SilentServicesWYF, 2023-01-30:
hello I have same first issue with you, but I can't find libsdrpp_core.so in /usr/lib, can you please help me?

Patronics, 2024-11-13:
I just wanted to report in case it's helpful for anyone else, I had the same issue on Asahi Fedora 40, and the solution of symlinking the file to lib64 with the following command worked perfectly for me.

`sudo ln -s /usr/lib/libsdrpp_core.so /usr/lib64/`