AlexandreRouma/SDRPlusPlus #396: Couldn't load ./modules/audio_sink.dll. Error code: 126

bkram, 2021-10-01:
For some reason the audio_sink.dll cannot be loaded, it does exist in the modules directory.

Any clue what is going wrong ?

Microsoft Windows NT 10.0.19043.0 

Start with logging in console.
```
[2021-10-01 14:03:40.519] [info] SDR++ v1.0.3
[2021-10-01 14:03:40.520] [info] Loading config
[2021-10-01 14:03:40.953] [info] Loading icons
[2021-10-01 14:03:40.969] [info] Loading band plans
[2021-10-01 14:03:40.972] [info] Loading band plans color table
[2021-10-01 14:03:40.978] [info] Loading modules
[2021-10-01 14:03:40.978] [info] Loading ./modules/airspyhf_source.dll
[2021-10-01 14:03:40.982] [info] Loading ./modules/airspy_source.dll
[2021-10-01 14:03:40.984] [info] Loading ./modules/audio_sink.dll
[2021-10-01 14:03:40.987] [error] Couldn't load ./modules/audio_sink.dll. Error code: 126
[2021-10-01 14:03:40.987] [info] Loading ./modules/bladerf_source.dll
[2021-10-01 14:03:40.991] [info] Loading ./modules/discord_integration.dll
[2021-10-01 14:03:40.993] [info] Loading ./modules/file_source.dll
[2021-10-01 14:03:40.996] [info] Loading ./modules/frequency_manager.dll
[2021-10-01 14:03:40.998] [info] Loading ./modules/hackrf_source.dll
[2021-10-01 14:03:41.002] [info] Loading ./modules/limesdr_source.dll
[2021-10-01 14:03:41.005] [info] Loading ./modules/meteor_demodulator.dll
[2021-10-01 14:03:41.007] [info] Loading ./modules/network_sink.dll
[2021-10-01 14:03:41.010] [info] Loading ./modules/plutosdr_source.dll
[2021-10-01 14:03:41.014] [info] Loading ./modules/radio.dll
[2021-10-01 14:03:41.017] [info] Loading ./modules/recorder.dll
[2021-10-01 14:03:41.019] [info] Loading ./modules/rigctl_server.dll
[2021-10-01 14:03:41.022] [info] Loading ./modules/rtl_sdr_source.dll
[2021-10-01 14:03:41.025] [info] Loading ./modules/rtl_tcp_source.dll
[2021-10-01 14:03:41.028] [info] Loading ./modules/sdrplay_source.dll
[2021-10-01 14:03:41.033] [info] Loading ./modules/spyserver_source.dll
[2021-10-01 14:03:41.035] [info] Initializing Airspy Source (airspy_source)
[2021-10-01 14:03:41.051] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-10-01 14:03:41.064] [info] Initializing Audio Sink (audio_sink)
[2021-10-01 14:03:41.065] [error] Module 'audio_sink' doesn't exist
[2021-10-01 14:03:41.065] [info] Initializing BladeRF Source (bladerf_source)
[2021-10-01 14:03:41.079] [error] Could not list devices
[2021-10-01 14:03:41.079] [info] Initializing File Source (file_source)
[2021-10-01 14:03:41.080] [info] Initializing Frequency Manager (frequency_manager)
[2021-10-01 14:03:41.081] [info] Initializing HackRF Source (hackrf_source)
[2021-10-01 14:03:41.097] [info] Initializing LimeSDR Source (limesdr_source)
[2021-10-01 14:03:41.098] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-10-01 14:03:41.098] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-10-01 14:03:41.113] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-10-01 14:03:41.113] [info] Initializing Radio (radio)
[2021-10-01 14:03:41.140] [info] Initializing Recorder (recorder)
[2021-10-01 14:03:41.141] [info] Initializing Rigctl Server (rigctl_server)
[2021-10-01 14:03:41.141] [info] Initializing SDRplay Source (sdrplay_source)
[2021-10-01 14:03:41.601] [info] New DSP samplerate: 10000000.0 (source samplerate is 10000000.0)
[2021-10-01 14:03:41.601] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.321] [info] Initializing SoapySDR Source (soapy_source)
[2021-10-01 14:03:42.322] [error] Module 'soapy_source' doesn't exist
[2021-10-01 14:03:42.322] [info] Initializing SpyServer Source (spyserver_source)
[2021-10-01 14:03:42.322] [info] Loading color maps
[2021-10-01 14:03:42.324] [info] Loading ./res/colormaps/classic.json
[2021-10-01 14:03:42.327] [info] Loading ./res/colormaps/electric.json
[2021-10-01 14:03:42.329] [info] Loading ./res/colormaps/gqrx.json
[2021-10-01 14:03:42.332] [info] Loading ./res/colormaps/greyscale.json
[2021-10-01 14:03:42.334] [info] Loading ./res/colormaps/inferno.json
[2021-10-01 14:03:42.336] [info] Loading ./res/colormaps/magma.json
[2021-10-01 14:03:42.338] [info] Loading ./res/colormaps/plasma.json
[2021-10-01 14:03:42.340] [info] Loading ./res/colormaps/turbo.json
[2021-10-01 14:03:42.342] [info] Loading ./res/colormaps/viridis.json
[2021-10-01 14:03:42.347] [info] Loading ./res/colormaps/websdr.json
[2021-10-01 14:03:42.365] [info] New DSP samplerate: 10000000.0 (source samplerate is 10000000.0)
[2021-10-01 14:03:42.365] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.366] [info] SDRPlaySourceModule 'SDRplay Source': Menu Select!
[2021-10-01 14:03:42.367] [info] New DSP samplerate: 10000000.0 (source samplerate is 10000000.0)
[2021-10-01 14:03:42.368] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.387] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.387] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.389] [info] Updating FFT DSP settings: Keep: 65536, Skip: 434464
[2021-10-01 14:03:42.390] [info] SDRPlaySourceModule 'SDRplay Source': Tune: 104100000.0!
[2021-10-01 14:03:42.390] [info] Running post-init for Airspy Source
[2021-10-01 14:03:42.390] [info] Running post-init for AirspyHF+ Source
[2021-10-01 14:03:42.390] [info] Running post-init for BladeRF Source
[2021-10-01 14:03:42.390] [info] Running post-init for File Source
[2021-10-01 14:03:42.390] [info] Running post-init for Frequency Manager
[2021-10-01 14:03:42.390] [info] Running post-init for HackRF Source
[2021-10-01 14:03:42.390] [info] Running post-init for LimeSDR Source
[2021-10-01 14:03:42.390] [info] Running post-init for PlutoSDR Source
[2021-10-01 14:03:42.390] [info] Running post-init for RTL-SDR Source
[2021-10-01 14:03:42.390] [info] Running post-init for RTL-TCP Source
[2021-10-01 14:03:42.390] [info] Running post-init for Radio
[2021-10-01 14:03:42.390] [info] Running post-init for Recorder
[2021-10-01 14:03:42.390] [info] Running post-init for Rigctl Server
[2021-10-01 14:03:42.390] [info] Running post-init for SDRplay Source
[2021-10-01 14:03:42.390] [info] Running post-init for SpyServer Source
[2021-10-01 14:03:42.390] [info] Ready.
```

gvanem, 2021-10-01:
> Any clue what is going wrong ?

Code 126 is `ERROR_MOD_NOT_FOUND` and `audio_sink.dll` normally depends on `rtaudio.dll`. Is it there or on `PATH`?

AlexandreRouma, 2021-10-01:
rtaudio should be included in the build

bkram, 2021-10-05:
The rtaudio.ddl is included in the directory, but for some reason the error is now gone, I expect that for some reason installing VB-Audio Virtual Cable apparently fixed the issue

closed 2021-10-05 by bkram