AlexandreRouma/SDRPlusPlus #397: Negative frequencies from .WAV-file

gvanem, 2021-10-01:
I used the `file_source.dll` to play an old .wav-file recording (32-bit, from 2009). I was surprised that it showed the spectrum mirrored:
[image]

Comparing with SDR-Sharp v1772, I see no negative frequencies:
[image]

Is this *working as designed*?

AlexandreRouma, 2021-10-01:
the "negative" frequencies is because it can't find the center frequency in the filename so it just default to zero.

For the mirroring however I'm not sure what's going on without looking a the wav file, could you upload it somewhere?

gvanem, 2021-10-01:
Here: http://www.watt-32.net/dx/PR_090226-0558_01200kHz_000_055900_060030.wav

AlexandreRouma, 2021-10-01:
This file looks  nothing like the first screenshot on my install

AlexandreRouma, 2021-10-01:
I'll check the wav file itself, it's behaving a bit strangely

AlexandreRouma, 2021-10-01:
Ok found, the issue, looks like old SDR# versions added a bunch of junk to the beginning of the file, removing it lets it play correctly
[image]

gvanem, 2021-10-01:
> This file looks nothing like the first screenshot on my install

Try zooming a bit. The file I uploaded **is** the same .wav-file in this PR.

> looks like old SDR# versions added a bunch of junk to the beginning of the file

You mean that this `PR_090226-0558_01200kHz_000_055900_060030.wav` from 2009 was recorded with SDR-Sharp? 
Hard to believe since the 1st SDR-Sharp was released around 2013.

AlexandreRouma, 2021-10-01:
Oh alright I thought it was SDR#
Well whatever recorded it added some metadata in front of the audio data which causes it to read the sames wrong

closed 2022-09-29 by AlexandreRouma

bczhc, 2024-07-13:
Met the same thing, and then reached here. Just for the information, one way to remove these metadata is via:
```bash
ffmpeg -i input.wav -fflags +bitexact -flags:v +bitexact -flags:a +bitexact output.wav
```

Then SDR++ will handle it correctly.